ebroecker/canmatrix #509: ValueError: max() arg is an empty sequence

nostar, 2020-09-27:
FYI this is running the version with my change in issue #508 in place.  DBF files for 8 out of the 13 clusters are generated before the crash.

```
$ canconvert --deleteZeroSignals --cutLongFrames=8 file.arxml file.dbf
INFO - convert - Importing file.arxml ... 
INFO - canmatrix - max calculation for 136 not possible using 128 as base for max value
INFO - canmatrix - max calculation for 136 not possible using 128 as base for max value
INFO - canmatrix - max calculation for 136 not possible using 128 as base for max value
INFO - canmatrix - max calculation for 136 not possible using 128 as base for max value
INFO - canmatrix - max calculation for 136 not possible using 128 as base for max value
INFO - canmatrix - max calculation for 136 not possible using 128 as base for max value
INFO - canmatrix - max calculation for 136 not possible using 128 as base for max value
INFO - canmatrix - max calculation for 136 not possible using 128 as base for max value
INFO - convert - done
INFO - convert - Exporting file.dbf ... 
INFO - convert - Cluster1
INFO - convert - 7 Frames found
INFO - convert - Cluster2
INFO - convert - 2 Frames found
INFO - convert - Cluster3
INFO - convert - 566 Frames found
INFO - convert - Cluster4
INFO - convert - 1233 Frames found
INFO - convert - Cluster5
INFO - convert - 234 Frames found
INFO - convert - Cluster6
INFO - convert - 401 Frames found
INFO - convert - Cluster7
INFO - convert - 1298 Frames found
INFO - convert - Cluster8
INFO - convert - 905 Frames found
INFO - convert - Cluster9
INFO - convert - 850 Frames found
INFO - convert - Cluster10
INFO - convert - 540 Frames found
INFO - convert - Cluster11
INFO - convert - 6 Frames found
INFO - convert - Cluster12
INFO - convert - 3 Frames found
INFO - convert - Cluster13
INFO - convert - 62 Frames found
Traceback (most recent call last):
  File "/usr/local/bin/canconvert", line 11, in <module>
    load_entry_point('canmatrix===0caf5fe.post11.dev0-gc6a24f3', 'console_scripts', 'canconvert')()
  File "/usr/local/lib/python3.8/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.8/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.8/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.8/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python3.8/site-packages/canmatrix-0caf5fe.post11.dev0_gc6a24f3-py3.8.egg/canmatrix/cli/convert.py", line 135, in cli_convert
    canmatrix.convert.convert(infile, outfile, **options)
  File "/usr/local/lib/python3.8/site-packages/canmatrix-0caf5fe.post11.dev0_gc6a24f3-py3.8.egg/canmatrix/convert.py", line 214, in convert
    canmatrix.formats.dumpp(out_dbs, out_file_name, **options)
  File "/usr/local/lib/python3.8/site-packages/canmatrix-0caf5fe.post11.dev0_gc6a24f3-py3.8.egg/canmatrix/formats/__init__.py", line 135, in dumpp
    dump(db, file_object, export_type, **options)
  File "/usr/local/lib/python3.8/site-packages/canmatrix-0caf5fe.post11.dev0_gc6a24f3-py3.8.egg/canmatrix/formats/__init__.py", line 110, in dump
    module_instance.dump(can_matrix_or_cluster, file_object, **options)  # type: ignore
  File "/usr/local/lib/python3.8/site-packages/canmatrix-0caf5fe.post11.dev0_gc6a24f3-py3.8.egg/canmatrix/formats/dbf.py", line 340, in dump
    if max([x.cycle_time for y in db.frames for x in y.signals]) > 0:
ValueError: max() arg is an empty sequence
```

ebroecker, 2020-09-27:
I think, this error could is a consequence from your change in #508 

could your try again?

nostar, 2020-09-27:
I thought the same thing, but this issue remains with latest pull:
```
Traceback (most recent call last):
  File "/usr/local/bin/canconvert", line 11, in <module>
    load_entry_point('canmatrix===0caf5fe.post12-g81eac11', 'console_scripts', 'canconvert')()
  File "/usr/local/lib/python3.8/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.8/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.8/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.8/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python3.8/site-packages/canmatrix-0caf5fe.post12_g81eac11-py3.8.egg/canmatrix/cli/convert.py", line 135, in cli_convert
    canmatrix.convert.convert(infile, outfile, **options)
  File "/usr/local/lib/python3.8/site-packages/canmatrix-0caf5fe.post12_g81eac11-py3.8.egg/canmatrix/convert.py", line 214, in convert
    canmatrix.formats.dumpp(out_dbs, out_file_name, **options)
  File "/usr/local/lib/python3.8/site-packages/canmatrix-0caf5fe.post12_g81eac11-py3.8.egg/canmatrix/formats/__init__.py", line 135, in dumpp
    dump(db, file_object, export_type, **options)
  File "/usr/local/lib/python3.8/site-packages/canmatrix-0caf5fe.post12_g81eac11-py3.8.egg/canmatrix/formats/__init__.py", line 110, in dump
    module_instance.dump(can_matrix_or_cluster, file_object, **options)  # type: ignore
  File "/usr/local/lib/python3.8/site-packages/canmatrix-0caf5fe.post12_g81eac11-py3.8.egg/canmatrix/formats/dbf.py", line 340, in dump
    if max([x.cycle_time for y in db.frames for x in y.signals]) > 0:
ValueError: max() arg is an empty sequence
```

ebroecker, 2020-09-27:
could you try branch iss509?

https://github.com/ebroecker/canmatrix/tree/iss509

nostar, 2020-09-27:
Same results as last time:
```
Traceback (most recent call last):
  File "/usr/local/bin/canconvert", line 11, in <module>
    load_entry_point('canmatrix===0caf5fe.post12-g81eac11', 'console_scripts', 'canconvert')()
  File "/usr/local/lib/python3.8/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.8/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.8/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.8/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python3.8/site-packages/canmatrix-0caf5fe.post12_g81eac11-py3.8.egg/canmatrix/cli/convert.py", line 135, in cli_convert
    canmatrix.convert.convert(infile, outfile, **options)
  File "/usr/local/lib/python3.8/site-packages/canmatrix-0caf5fe.post12_g81eac11-py3.8.egg/canmatrix/convert.py", line 214, in convert
    canmatrix.formats.dumpp(out_dbs, out_file_name, **options)
  File "/usr/local/lib/python3.8/site-packages/canmatrix-0caf5fe.post12_g81eac11-py3.8.egg/canmatrix/formats/__init__.py", line 135, in dumpp
    dump(db, file_object, export_type, **options)
  File "/usr/local/lib/python3.8/site-packages/canmatrix-0caf5fe.post12_g81eac11-py3.8.egg/canmatrix/formats/__init__.py", line 110, in dump
    module_instance.dump(can_matrix_or_cluster, file_object, **options)  # type: ignore
  File "/usr/local/lib/python3.8/site-packages/canmatrix-0caf5fe.post12_g81eac11-py3.8.egg/canmatrix/formats/dbf.py", line 340, in dump
    if max([x.cycle_time for y in db.frames for x in y.signals]) > 0:
ValueError: max() arg is an empty sequence
```
I confirmed that this run was executing the right install.  This branch was installed to /usr/local/lib/python3.8/site-packages/canmatrix-0caf5fe.post12_g81eac11-py3.8.egg and the development branch to /usr/local/lib/python3.8/site-packages/canmatrix-0caf5fe.post11.dev0_gc6a24f3-py3.8.egg

-Doug

ebroecker, 2020-09-28:
you are right, I did forget one point to fix,
Now it should work

nostar, 2020-09-28:
Yes this is fixed now in iss509.

closed 2020-09-28 by ebroecker