ebroecker/canmatrix #515: Feature request: ARXML -> LDF

nostar, 2020-09-30:
I see you have started work on other cluster types (ethernet, flexray, etc).  Another nice feature would be ability to export LIN clusters from ARXML to LDF.

svanimisetti, 2022-03-20:
Looking for same feature - but not sure if it was implemented? If I use the latest development release, I get an error stating "*This file format is not supported for writing*". If it is still in the works, what can I do to contribute/help towards implementation this feature?

ebroecker, 2022-03-21:
Hi @svanimisetti,

no - it is not implemented, not even started to implement.
I don't have enough knowledge about lin.

Parsing the relevant information from ARXML should not be to hard, anyway.

If you provide some example we could starte some basic implementation.

ebroecker, 2023-07-18:
Any examples ? Anything new? 
I'll start implementing if I get some examples... (also possible as PN)

ebroecker, 2024-09-19:
please reopen if input available