ebroecker/canmatrix #520: Cannot convert from .xlsx to dbc

Ajay-Embed, 2020-10-20:
[image]

Getting this error. Any idea why?

ebroecker, 2020-10-20:
Hi  @Ajay-Embed 

did you use "cmTemplate.xlsx"

There should be a column "signalSNA"

Ajay-Embed, 2020-10-20:
Hello, Yes I created a column by this name but still the error persists.

ebroecker, 2020-10-20:
Hi  @Ajay-Embed 

could you try "pure" cmTemplate.xlsx?

Ajay-Embed, 2020-10-20:
Yes, that file is working, But my custom xlsx file is sending out the above posted error even after adding the signalSNA column..

ebroecker, 2020-10-20:
there must be some typo, white-space whatever which confuses the xls-importer

Ajay-Embed, 2020-10-21:
Hi @ebroecker,  The script is also not able to parse 8 byte CAN msg ID's... For example, In place of 123h, I passed 0x1BFF0989. It shows msg ID too long.

ebroecker, 2020-10-21:
Hi @Ajay-Embed ,

you have to tell canmatrix, that you are dealing with an extended ID, thus you have to write 1BFF0989**x**h.

closed 2020-10-22 by Ajay-Embed