ebroecker/canmatrix #521: Imports error on Python 3.8.3

fskoras, 2020-10-20:
I have freshly installed canmatrix with command

pip install canmatrix

when I tried to import the module I got the below error message

>>> import canmatrix
arxml is not supported
Traceback (most recent call last):
  File "D:\Tools\Python\Python38\lib\site-packages\canmatrix\formats\__init__.py", line 27, in <module>
    importlib.import_module("canmatrix.formats." + module)
  File "D:\Tools\Python\Python38\lib\importlib\__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "D:\Tools\Python\Python38\lib\site-packages\canmatrix\formats\arxml.py", line 34, in <module>
    import lxml.etree
ModuleNotFoundError: No module named 'lxml'
kcd is not supported
Traceback (most recent call last):
  File "D:\Tools\Python\Python38\lib\site-packages\canmatrix\formats\__init__.py", line 27, in <module>
    importlib.import_module("canmatrix.formats." + module)
  File "D:\Tools\Python\Python38\lib\importlib\__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "D:\Tools\Python\Python38\lib\site-packages\canmatrix\formats\kcd.py", line 35, in <module>
    import lxml.etree
ModuleNotFoundError: No module named 'lxml'
fibex is not supported
Traceback (most recent call last):
  File "D:\Tools\Python\Python38\lib\site-packages\canmatrix\formats\__init__.py", line 27, in <module>
    importlib.import_module("canmatrix.formats." + module)
  File "D:\Tools\Python\Python38\lib\importlib\__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "D:\Tools\Python\Python38\lib\site-packages\canmatrix\formats\fibex.py", line 33, in <module>
    import lxml.etree
ModuleNotFoundError: No module named 'lxml'
xls is not supported
Traceback (most recent call last):
  File "D:\Tools\Python\Python38\lib\site-packages\canmatrix\formats\__init__.py", line 27, in <module>
    importlib.import_module("canmatrix.formats." + module)
  File "D:\Tools\Python\Python38\lib\importlib\__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "D:\Tools\Python\Python38\lib\site-packages\canmatrix\formats\xls.py", line 34, in <module>
    import xlrd
ModuleNotFoundError: No module named 'xlrd'
xlsx is not supported
Traceback (most recent call last):
  File "D:\Tools\Python\Python38\lib\site-packages\canmatrix\formats\__init__.py", line 27, in <module>
    importlib.import_module("canmatrix.formats." + module)
  File "D:\Tools\Python\Python38\lib\importlib\__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "D:\Tools\Python\Python38\lib\site-packages\canmatrix\formats\xlsx.py", line 32, in <module>
    import xlsxwriter
ModuleNotFoundError: No module named 'xlsxwriter'
yaml is not supported
Traceback (most recent call last):
  File "D:\Tools\Python\Python38\lib\site-packages\canmatrix\formats\__init__.py", line 27, in <module>
    importlib.import_module("canmatrix.formats." + module)
  File "D:\Tools\Python\Python38\lib\importlib\__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "D:\Tools\Python\Python38\lib\site-packages\canmatrix\formats\yaml.py", line 32, in <module>
    import yaml
ModuleNotFoundError: No module named 'yaml'

ebroecker, 2020-10-20:
Hi,

you need to install the requirements also.
https://canmatrix.readthedocs.io/en/latest/installation.html

pip install lxml, xlrd, ....

fskoras, 2020-10-20:
That worked thanks.

closed 2020-10-20 by fskoras