ebroecker/canmatrix #522: Arxml parser 'Frame' object: 'receivers' property is always an empty list

fskoras, 2020-10-22:
I have manually explored my file in Vector AUTOSAR System Description Network Explorer PRO program and it shows correctly the Receivers filed. However when I try to access this information from within canmatrix 'Frame' property 'receivers' it is empty for all the frames. 'Signal' objects show receivers correctly.

ebroecker, 2020-10-22:
well - ok - this is the dbc-history of canmatrix.
receivers in dbc are only defined in signals. 
Implicite it is clear, that frames are also received from the same ecus.
It's easy to fix.

fskoras, 2020-10-23:
It makes sense to me, but the arxml format allows you to create files in which the frame has a specific receiver and the signal connected to them has no specific receivers. This was the case with my file.

ebroecker, 2020-10-23:
yes - I'll fix this the next days

ebroecker, 2020-10-30:
could you please try branch 522?

pip install git+https://github.com/ebroecker/canmatrix.git@iss522

fskoras, 2020-11-02:
Thanks. It is working now for frames as well. I have one more observation. Transmitters seems to be having the same issue

ebroecker, 2021-05-07:
Hi @fskoras 
is this still an issue ?

closed 2022-03-31 by ebroecker