ebroecker/canmatrix #523: Errors on import

rfrowe, 2020-10-30:
This is an extension of #521.

`setup.py` defines various extras for each format. For example, one could do `pip install cantools[arxml]` which would add an additional dependency on `lxml`.

Right now, the package must be installed as `pip install cantools[arxml,xls,xlsx,yaml]` otherwise errors will be printed on import. I don't like an unsuppressable error message for dependencies that I don't want nor need. Right now, you've allowed for clients to install with extra dependencies, which implies that the core package should work with those "extras" not installed.

It is true that the package still _works_, but please either change your logging logic to allow clients to set the logging level or make a change such that `import cantools.formats` causes the logs but `import cantools.formats.dbc` doesn't. I only use `cantools.formats.dbc` and its the only import I make, so I don't care about whether arxml is supported because I don't support it.

ebroecker, 2023-07-18:
Hi @rfrowe 

I think the errors should be logger output since https://github.com/ebroecker/canmatrix/commit/eecffcc34411c8533a00cedb154a34f03fd72082 - so can we close this issue?

ebroecker, 2023-07-19:
Hi @rfrowe 

Thanks a lot for your quick response!

rfrowe, 2023-07-25:
Np, glad to see this fixed nonetheless :)