ebroecker/canmatrix #531: Feature request: LIN bus *.LDF to *.DBC format

MatinF, 2020-12-02:
Hi Eduard, 

Would it be possible to support loading a LIN bus *.LDF file into canmatrix - with the use case of e.g. creating a DBC file from the loaded LDF file?

This would e.g. allow tools like asammdf that rely on canmatrix to let users load *.LDF files directly to decode LIN bus data - or to alternatively use canmatrix to first create a DBC file from their LDF files. 

We would be able to provide sample LIN bus data in the MDF4 format as well as LDF files for this purpose via e-mail. Further, we would be happy to work with you to test and evaluate the code. 

Let me know if this is something you'd be open to.

danielhrisca, 2020-12-03:
It would be great to load the LDF files as canmatrix.Matrix objects as well

reymor, 2021-01-17:
@MatinF @ebroecker 

Hello, 

I'm open to contribute to this new feature. Let me know how I could help you.

ebroecker, 2021-01-17:
using ldfparser (https://github.com/c4deszes/ldfparser) looks very promising to me.

Gonna hack something thus we have a base to discuss.

ebroecker, 2021-01-23:
very simply support in:
https://github.com/ebroecker/canmatrix/tree/iss531_import_ldf

https://github.com/ebroecker/canmatrix/blob/iss531_import_ldf/src/canmatrix/formats/ldf.py

maybe you can have a look, ...

MatinF, 2021-01-23:
Awesome Eduard. Daniel if you get the chance to test adding this to asammdf I can do some tests

MatinF, 2021-02-09:
Hi Eduard, would you be able to include this functionality in an upcoming release? Then it may be possible to include it in e.g. asammdf and our Python API.

ebroecker, 2021-02-09:
Does it fullfill your needs?
Did you test?

If it works for your needs certainly I'll add it to a future release

danielhrisca, 2021-02-10:
I'm getting this error when I try to load a LDF file:

```
File "D:\02__PythonWorkspace\_venv\lib\site-packages\canmatrix\formats\__init__.py", line 96, in load
    dbs[key] = module_instance.load(file_object, **options)  # type: ignore
  File "D:\02__PythonWorkspace\_venv\lib\site-packages\canmatrix\formats\ldf.py", line 7, in load
    ldf = ldfparser.parseLDF(path=f.name)  # using f.name is not nice, but works
AttributeError: '_io.BytesIO' object has no attribute 'name'

```

danielhrisca, 2021-02-10:
I've managed to load the LDF file only by using `canmatrix.formats.loadp` instead of `canmatrix.formats.loads`

Here are the first missing items:
* conversion rules are not loaded
* I don't think that there is an explicit signed/unsigned declaration in the ldf so this would have to be deduced from the conversion definition (there it has the min and max raw range)

MatinF, 2021-02-10:
It should also be noted perhaps that some LIN messages contain signals that have differing decoding rules depending on the physical value. For example the below message:

```
  Enc_BatteryCurrent {
    physical_value, 1, 66, 10, -1230, "Ampere" ;
    physical_value, 67, 4765, 0.1, -576.6, "Ampere" ;
    physical_value, 4766, 12765, 0.01, -147.66, "Ampere" ;
    physical_value, 12766, 52765, 0.001, -32.766, "Ampere" ;
    physical_value, 52766, 60765, 0.01, -507.66, "Ampere" ;
    physical_value, 60766, 65465, 0.1, -5976.6, "Ampere" ;
    physical_value, 65466, 65532, 10, -654090, "Ampere" ;
    logical_value, 0, "IBatt_lowerlimit" ;
    logical_value, 65533, "IBatt_upperlimit" ;
    logical_value, 65534, "BatteryCurrent_Error" ;
    logical_value, 65535, "LIN_INIT" ;
  }
```

In such cases, the decoding rule logic cannot be fully handled within the DBC framework. My recommendation for now would be to simply ignore such messages if possible.

ebroecker, 2021-02-10:
Thanks for the feedback, gonna work on it

MatinF, 2021-02-12:
In regards to comparing LDF vs DBC, you might also find the below article section useful Eduard:
https://www.csselectronics.com/screen/page/lin-bus-protocol-intro-basics/language/en#lin-description-file

Martin

ebroecker, 2021-02-21:
did someone have a chance to check https://github.com/ebroecker/canmatrix/commit/acdd5d3e7f012c43d8434b6590614ffb5f8c00fc ?

MatinF, 2021-02-21:
@danielhrisca not sure if this is the version you included in the asammdf dev version I tested

danielhrisca, 2021-03-07:
@ebroecker the new commit throws this exception in `calc_max`

```
File "D:\02__PythonWorkspace\_venv\lib\site-packages\canmatrix\canmatrix.py", line 395, in calc_max
    return self.offset + (self.float_factory(rawMax) * self.factor)
TypeError: unsupported operand type(s) for *: 'decimal.Decimal' and 'float'

```

ebroecker, 2022-03-30:
I'll close this issue, please open new issues for problems with current implementation

MatinF, 2024-12-10:
Hi Edward,

I would like to test this out to e.g. see if I can load attached LDF file and export it to DBC. I tried using below script, but I get this error:

```
import canmatrix.formats
import sys

def convert_ldf_to_dbc(ldf_file: str, dbc_file: str):
    """
    Converts an LDF (LIN Description File) to a DBC (CAN Database File).
    
    Parameters:
        ldf_file (str): Path to the input LDF file.
        dbc_file (str): Path to the output DBC file.
    """
    try:
        # Load the LDF file
        print(f"Loading LDF file: {ldf_file}")
        matrix = canmatrix.formats.loadp(ldf_file)
        
        # Check if the loading was successful
        if matrix is None:
            raise ValueError("Failed to parse the LDF file.")
        
        print(f"Successfully loaded LDF file: {ldf_file}")
        
        # Export the matrix to a DBC file
        print(f"Exporting to DBC file: {dbc_file}")
        canmatrix.formats.dump(matrix, dbc_file)
        
        print(f"Successfully exported to DBC file: {dbc_file}")
    except Exception as e:
        print(f"Error during conversion: {e}")

if __name__ == "__main__":
    # Example usage: python convert_ldf_to_dbc.py input.ldf output.dbc
    if len(sys.argv) != 3:
        print("Usage: python convert_ldf_to_dbc.py <input_ldf_file> <output_dbc_file>")
        sys.exit(1)
    
    input_ldf = sys.argv[1]
    output_dbc = sys.argv[2]
    convert_ldf_to_dbc(input_ldf, output_dbc)

```

Here is the script:

```
import canmatrix.formats
import sys

def convert_ldf_to_dbc(ldf_file: str, dbc_file: str):
    """
    Converts an LDF (LIN Description File) to a DBC (CAN Database File).
    
    Parameters:
        ldf_file (str): Path to the input LDF file.
        dbc_file (str): Path to the output DBC file.
    """
    try:
        # Load the LDF file
        print(f"Loading LDF file: {ldf_file}")
        matrix = canmatrix.formats.loadp(ldf_file)
        
        # Check if the loading was successful
        if matrix is None:
            raise ValueError("Failed to parse the LDF file.")
        
        print(f"Successfully loaded LDF file: {ldf_file}")
        
        # Export the matrix to a DBC file
        print(f"Exporting to DBC file: {dbc_file}")
        canmatrix.formats.dump(matrix, dbc_file)
        
        print(f"Successfully exported to DBC file: {dbc_file}")
    except Exception as e:
        print(f"Error during conversion: {e}")

if __name__ == "__main__":
    # Example usage: python convert_ldf_to_dbc.py input.ldf output.dbc
    if len(sys.argv) != 3:
        print("Usage: python convert_ldf_to_dbc.py <input_ldf_file> <output_dbc_file>")
        sys.exit(1)
    
    input_ldf = sys.argv[1]
    output_dbc = sys.argv[2]
    convert_ldf_to_dbc(input_ldf, output_dbc)

```
The file I am trying to load is attached. Any idea what the issue may be?

[LIN_LDF_EXAMPLE.zip](https://github.com/user-attachments/files/18078632/LIN_LDF_EXAMPLE.zip)

reymor, 2024-12-10:
Hello @MatinF 

I think that you need to install `ldfparser` package

MatinF, 2024-12-10:
I tried updating the `dump()` function to take a 3rd input, `"dbc"`, which results in below:

```
C:\Users\marti\Downloads\LIN_LDF_EXAMPLE>python testy.py IBS_200X_P_24V.ldf test2.dbc
xls is not supported
xlsx is not supported
Loading LDF file: IBS_200X_P_24V.ldf
Successfully loaded LDF file: IBS_200X_P_24V.ldf
Exporting to DBC file: test2.dbc
Successfully exported to DBC file: test2.dbc
```

However, the file is not exported to the expected location that I provide. It seems to load correctly if I print the `matrix` object, so I guess the problem is related to my use of `canmatrix.formats.dump`.

reymor, 2024-12-10:
I think that it is because there is no 1to1 conversion between `dbc` and `ldf`. 

I understand that multiple entries for physical value is not something possible in `dbc` file format. However, 

> Would this be possible to introduce in the parsing of LDF files so that multiple physical value ranges get translated automatically into extended multiplexing ranges?

I think that it is possible

So, here let me understand what you mean:

```
  Enc_BatteryCurrent {
    physical_value, 1, 66, 10, -1230, "Ampere" ;
    physical_value, 67, 4765, 0.1, -576.6, "Ampere" ;
    physical_value, 4766, 12765, 0.01, -147.66, "Ampere" ;
    physical_value, 12766, 52765, 0.001, -32.766, "Ampere" ;
    physical_value, 52766, 60765, 0.01, -507.66, "Ampere" ;
    physical_value, 60766, 65465, 0.1, -5976.6, "Ampere" ;
    physical_value, 65466, 65532, 10, -654090, "Ampere" ;
    logical_value, 0, "IBatt_lowerlimit" ;
    logical_value, 65533, "IBatt_upperlimit" ;
    logical_value, 65534, "BatteryCurrent_Error" ;
    logical_value, 65535, "LIN_INIT" ;
  }
```

How do you use the `extended multiplexing` here?

MatinF, 2024-12-11:
I believe a 1:1 mapping can be achieved for most LIN signals, with the primary challenge being the cases where a single LIN signal has multiple physical value representations (as in the example I provided). My suggestion in this case would be to consider if this could be handled via extended multiplexing. I illustrate how this would work for the specific example of BatteryCurrent via the picture I shared from Kvaser's DBC editor. Here I use the 16 bits reflecting the BatteryCurrent range and I essentially use that as a multiplexor to modify the interpretation of the same 16 bit range for the purpose of calculating a BatteryCurrent1, BatteryCurrent2, ... signal. Essentially adding a suffix index for each unique physical_value entry of the signal. 

The resulting DBC works as intended in the sense that it will correctly DBC decode the data using the scaling/offset specified for the range defined in the multiplexing. 

Let me know if it makes more sense? I am not sure if it becomes too complex to create this form of conversion in canmatrix, though. Maybe a starting point could be that canmatrix is able to load the LDF and correctly map the various physical_value ranges correctly for the internal database object somehow - I am not sure if that is done currently. The hope would be to allow tools like the asammdf GUI to then properly load LDF files and use it for decoding MF4 files that contain LIN bus data out-the-box. If this could be achieved, it would to us be more relevant than e.g. being able to export a DBC equivalent format.

MatinF, 2024-12-11:
Looking at the internal format of the canmatrix database, I guess the fundamental issue is that - similar to DBC files - the internal format only expects a single scale/offset value per signal (although it is able to also incorporate the 'value table' provided by the logical values in the LDF). For example, the aforementioned `BatteryCurrent` signal seems to look as below in the internal format:

```
Frame(
    name="IBS_DataAcquisition",
    arbitration_id=ArbitrationId(
        id=33,
        extended=False
    ),
    size=8,
    transmitters=["IBS"],
    is_complex_multiplexed=False,
    is_fd=False,
    comment="",
    signals=[
        Signal(
            name="BatteryCurrent",
            start_bit=0,
            size=16,
            is_little_endian=True,
            is_signed=False,
            offset=-654090.0,
            factor=10.0,
            unit="Ampere",
            receivers=["SG"],
            comment=None,
            multiplex=None,
            mux_value=None,
            is_float=False,
            is_ascii=False,
            type_label="",
            enumeration=None,
            comments={},
            attributes={},
            values={
                0: "IBatt_lowerlimit",
                65533: "IBatt_upperlimit",
                65534: "BatteryCurrent_Error",
                65535: "LIN_INIT"
            },
            mux_val_grp=[],
            muxer_for_signal=None,
            calc_min_for_none=True,
            calc_max_for_none=True,
            cycle_time=0,
            initial_value=65535,
            min=65466,
            max=65532
        ),
        ...
    ]
)

```

As evident, the factor/offset corresponds to the last `physical_value` entry in the signal description of the LDF:

```
offset=-654090.0,
factor=10.0,
```

I guess that in order to properly resolve this, a rather fundamental structural change would be required in the canmatrix internal database format to handle multiple possible physical value entries per signal. One option might be to indeed leverage extended multiplexing as described above, as this seems to be supported by canmatrix from what I can tell. This would have the benefit that it would allow a direct export to DBC files as well. But it has the downside that a single signal, `BatteryCurrent` would have to be mapped to multiple signals, e.g. `BatteryCurrent1`, `BatteryCurrent2`, ... to reflect the full range of physical values described in the LDF. 

@ebroecker would be interested in your view on this and the feasibility of introducing this aspect of the LDF format into the canmatrix database format.

ebroecker, 2024-12-14:
For now we have only valuetables (1:1 Value to Label mapping) and scaling with factor and offset.
Anyway in LDF but also in ARXML it is possible to have different scales for different ranges. 

I think it could be the right thing to do to implement something that supports these more flexible raw to physical mappings.

ebroecker, 2024-12-18:
I added a simple change ,

https://github.com/ebroecker/canmatrix/pull/823

feel free to discuss

MatinF, 2025-01-14:
@ebroecker Will it be possible to merge this change into the master branch and release this as an update to the canmatrix repository?

ebroecker, 2025-02-10:
release 1.1 should be done

danielhrisca, 2025-02-11:
@ebroecker @MatinF 

There is a bug in the release 1.1

The signal `length` incoorectly has the multiplexer set to `ParameterID_Service01` when the DCBC found [here](https://github.com/danielhrisca/asammdf/files/4328945/OBD2-DBC-MDF4.zip) is loaded

### canamtrix 1.0

[image]

### canmatrix 1.1

[image]

### vector candb editor

[image]

MatinF, 2025-02-11:
Thanks for spotting this Daniel, appreciated!

@ebroecker Great if this can be patched in an updated release. Let me know if I can support on any testing.

ebroecker, 2025-02-12:
Hi @MatinF  @danielhrisca 

could you test the fix (already in development branch)?

danielhrisca, 2025-02-12:
I've just tested and it seems to work 👍

MatinF, 2025-02-14:
Hi @danielhrisca , 

If it's possible to create an updated development branch that resolves the LDF issue above that would be great. 

Unfortunately the version of asammdf we now host has the multiplex issue you reported. This is resolved in the latest dev version, but here the LDF functionality does not work - so it would be really cool to get an updated build that works with both items if possible :-)

ebroecker, 2025-02-15:
Hi @MatinF 
do you have some testcase fot the multiplex-issue?
Best thing would be we develope an unit-test for this issue, then I can be sure that it will be fixed in further releases...

MatinF, 2025-02-15:
I've sent you an MF4 and DBC that replicates the multiplexing issue that Daniel reported on an earlier version of asammdf that used canmatrix 1.1. But I think this has been resolved in the most recent version of asammdf development branch.

The problem for us is that the most recent asammdf development branch has re-introduced an issue with the LDF decoding where the wrong units and wrong scaling factors are used. So I am hoping to get that resolved in an update to the development branch of asammdf as well so we can host an updated version.

ebroecker, 2025-02-16:
Hi @MatinF,

meanwhile this discussion here is the showstopper of 1.2 (where the multiplex-issue is fixed).
I'm just waiting here for a "go" to release.

ebroecker, 2025-02-17:
Hi @MatinF 

Thank you for clearification!

ebroecker, 2025-02-17:
Just released 1.2

closed 2025-02-17 by ebroecker