ebroecker/canmatrix #533: arxml to dbc no output

manriqueply, 2020-12-03:
Im new to the canmatrix and im trying to convert a arxml database to dbc database, however, when doing it the exported file do not appear anywhere.

ebroecker, 2020-12-05:
well - it should.
It's named with the bus-name in the filename.

manriqueply, 2020-12-07:
Well.. its not.
The INFO says the convertion its done, however there's no other file in the folder, just the arxml file

ebroecker, 2020-12-09:
`python -m canmatrix.cli.convert -vv in.arxml out.dbc`

should create an output like

```
...
INFO - convert - Exporting out.dbc ... 
INFO - convert - BUS-NAME
INFO - convert - 29 Frames found
INFO - convert - done
```
in the folder you should have `out_BUS-NAME.dbc`

manriqueply, 2020-12-09:
I got this after trying

`INFO - convert - Importing Test.arxml ...
DEBUG - arxml - Read arxml ...
DEBUG - arxml - Build arTree ...
DEBUG - arxml - 0 frames in arxml...
DEBUG - arxml - 0 can-frame-triggering in arxml...
DEBUG - arxml - 0 SIGNAL-TO-PDU-MAPPINGS in arxml...
DEBUG - arxml - 236 I-SIGNAL-TO-I-PDU-MAPPING in arxml...
INFO - convert - done

INFO - convert - Exporting Test.dbc ...
INFO - convert - done`

ebroecker, 2020-12-10:
your ARXML does not contain CAN information - looks at least so...

you could try:
--arxmlIgnoreClusterInfo

or maybe:
--arxmlFlexray or --arxmlEthernet

closed 2021-05-07 by ebroecker