ebroecker/canmatrix #534: Following lines occur when converting from arxml to dbc

Qutadah, 2020-12-04:
Hello,

I am converting an arxml to dbc and the following lines occur by converting. I am not sure what they mean or if the conversion is falsed because these lines occur.

[image]

Thanks !

ebroecker, 2020-12-05:
Hi @Qutadah 
seems canmatrix does not like your compu-methods.
There are some options missing in the compu method decoding algorithm (maybe your min/max values are given "internal" but canmatrix expects "physical" for now...).

So maybe your min/max values are not converted correct.

ebroecker, 2022-03-31:
I close this, if it's still an issue, please shout!

closed 2022-03-31 by ebroecker