ebroecker/canmatrix #535: An error occurred during the conversion

xinkunlong, 2020-12-18:
Traceback (most recent call last):
  File "d:\kunlong\python\lib\runpy.py", line 193, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "d:\kunlong\python\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "D:\kunlong\python\Scripts\canconvert.exe\__main__.py", line 7, in <module>
  File "D:\kunlong\python\Lib\site-packages\click\core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "D:\kunlong\python\Lib\site-packages\click\core.py", line 782, in main
    rv = self.invoke(ctx)
  File "D:\kunlong\python\Lib\site-packages\click\core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "D:\kunlong\python\Lib\site-packages\click\core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "D:\kunlong\python\Lib\site-packages\canmatrix\cli\convert.py", line 135, in cli_convert
    canmatrix.convert.convert(infile, outfile, **options)
  File "D:\kunlong\python\Lib\site-packages\canmatrix\convert.py", line 40, in convert
    dbs = canmatrix.formats.loadp(infile, **options)
  File "D:\kunlong\python\Lib\site-packages\canmatrix\formats\__init__.py", line 77, in loadp
    return load(fileObject, import_type, key, **options)
  File "D:\kunlong\python\Lib\site-packages\canmatrix\formats\__init__.py", line 96, in load
    dbs[key] = module_instance.load(file_object, **options)  # type: ignore
  File "D:\kunlong\python\Lib\site-packages\canmatrix\formats\dbc.py", line 933, in load
    frame.cycle_time = int(frame.attributes.get("GenMsgCycleTime", 0))
ValueError: invalid literal for int() with base 10: '20.000'

ebroecker, 2020-12-20:
could you please try branch iss535?

pip install git+https://github.com/ebroecker/canmatrix.git@iss535

xinkunlong, 2020-12-21:
[image]
The problem has not been solved.

ebroecker, 2020-12-21:
Hi @xinkunlong 

something went wrong, if I look to https://github.com/ebroecker/canmatrix/blob/iss535/src/canmatrix/formats/dbc.py#L935 the code looks different...

ebroecker, 2021-05-07:
please open new issue if still a problem

closed 2021-05-07 by ebroecker