ebroecker/canmatrix #538: ModuleNotFoundError for format module imports with pyinstaller

leosh64, 2020-12-29:
In `formats/__init__.py` in [line 29](https://github.com/ebroecker/canmatrix/blob/568ed7d11b3875493ba3095826c4b48dcdea02e9/src/canmatrix/formats/__init__.py#L29), various format modules are imported, if they exist, and a warning is printed upon `ImportError`.

This works well on normal Python installations. However, when creating a stand-alone binary version of a software package that includes `canmatrix` with `pyinstaller,` the behavior changes: Instead of `ImportError`, a `ModuleNotFoundError` error is thrown, which is not caught in the current `__init__.py`:

```
csv is not supported
Traceback (most recent call last):
  File "canmatrix/formats/__init__.py", line 27, in <module>
  File "importlib/__init__.py", line 127, in import_module
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 973, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'canmatrix.formats.csv'
dbf is not supported
Traceback (most recent call last):
  File "canmatrix/formats/__init__.py", line 27, in <module>
  File "importlib/__init__.py", line 127, in import_module
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 973, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'canmatrix.formats.dbf'
json is not supported
Traceback (most recent call last):
  File "canmatrix/formats/__init__.py", line 27, in <module>
  File "importlib/__init__.py", line 127, in import_module
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 973, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'canmatrix.formats.json'
```

Perhaps `ModuleNotFoundError` should be caught as well.

leosh64, 2020-12-29:
Alternatively, the issue could probably also be addressed by adding pyinstaller `hooks` for `hiddenimports`: https://pyinstaller.readthedocs.io/en/stable/hooks.html#provide-hooks-with-package

danielhrisca, 2021-01-12:
I've created this PyInstaller issue https://github.com/pyinstaller/pyinstaller/issues/5467

Kochise, 2021-04-06:
Also present in a fresh install of Spyder 5.0.0 :

```
Python 3.7.9 (tags/v3.7.9:13c94747c7, Aug 17 2020, 18:58:18) [MSC v.1900 64 bit (AMD64)]
Type "copyright", "credits" or "license" for more information.

IPython 7.22.0 -- An enhanced Interactive Python.

In [1]: !pip install --upgrade canmatrix
Collecting canmatrix    ERROR: Command errored out with exit status 1:
     command: 'c:\program files\spyder\python\python.exe' -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\dkoch\\AppData\\Local\\Temp\\pip-install-v2qnl0qq\\future_1e2b2582703b4b05972e16b84744742d\\setup.py'"'"'; __file__='"'"'C:\\Users\\dkoch\\AppData\\Local\\Temp\\pip-install-v2qnl0qq\\future_1e2b2582703b4b05972e16b84744742d\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base 'C:\Users\dkoch\AppData\Local\Temp\pip-pip-egg-info-p4z30cn9'
         cwd: C:\Users\dkoch\AppData\Local\Temp\pip-install-v2qnl0qq\future_1e2b2582703b4b05972e16b84744742d\
    Complete output (5 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "C:\Users\dkoch\AppData\Local\Temp\pip-install-v2qnl0qq\future_1e2b2582703b4b05972e16b84744742d\setup.py", line 86, in <module>
        import src.future
    ModuleNotFoundError: No module named 'src'
    ----------------------------------------
WARNING: Discarding https://files.pythonhosted.org/packages/45/0b/38b06fd9b92dc2b68d58b75f900e97884c45bedd2ff83203d933cf5851c9/future-0.18.2.tar.gz#sha256=b1bead90b70cf6ec3f0710ae53a525360fa360d306a86583adc6bf83a4db537d (from https://pypi.org/simple/future/) (requires-python:>=2.6, !=3.0.*, !=3.1.*, !=3.2.*). Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
    ERROR: Command errored out with exit status 1:
     command: 'c:\program files\spyder\python\python.exe' -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\dkoch\\AppData\\Local\\Temp\\pip-install-v2qnl0qq\\future_4dff787b02d447e6ab8ea2989348e176\\setup.py'"'"'; __file__='"'"'C:\\Users\\dkoch\\AppData\\Local\\Temp\\pip-install-v2qnl0qq\\future_4dff787b02d447e6ab8ea2989348e176\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base 'C:\Users\dkoch\AppData\Local\Temp\pip-pip-egg-info-2ui_i4kf'
         cwd: C:\Users\dkoch\AppData\Local\Temp\pip-install-v2qnl0qq\future_4dff787b02d447e6ab8ea2989348e176\
    Complete output (5 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "C:\Users\dkoch\AppData\Local\Temp\pip-install-v2qnl0qq\future_4dff787b02d447e6ab8ea2989348e176\setup.py", line 86, in <module>
        import src.future
    ModuleNotFoundError: No module named 'src'
    ----------------------------------------
WARNING: Discarding https://files.pythonhosted.org/packages/3f/bf/57733d44afd0cf67580658507bd11d3ec629612d5e0e432beb4b8f6fbb04/future-0.18.1.tar.gz#sha256=858e38522e8fd0d3ce8f0c1feaf0603358e366d5403209674c7b617fa0c24093 (from https://pypi.org/simple/future/) (requires-python:>=2.6, !=3.0.*, !=3.1.*, !=3.2.*). Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
    ERROR: Command errored out with exit status 1:
     command: 'c:\program files\spyder\python\python.exe' -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\dkoch\\AppData\\Local\\Temp\\pip-install-v2qnl0qq\\future_1ec25e5865ec45149f3f11d75ded168d\\setup.py'"'"'; __file__='"'"'C:\\Users\\dkoch\\AppData\\Local\\Temp\\pip-install-v2qnl0qq\\future_1ec25e5865ec45149f3f11d75ded168d\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base 'C:\Users\dkoch\AppData\Local\Temp\pip-pip-egg-info-5zm5hd73'
         cwd: C:\Users\dkoch\AppData\Local\Temp\pip-install-v2qnl0qq\future_1ec25e5865ec45149f3f11d75ded168d\
    Complete output (5 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "C:\Users\dkoch\AppData\Local\Temp\pip-install-v2qnl0qq\future_1ec25e5865ec45149f3f11d75ded168d\setup.py", line 86, in <module>
        import src.future
    ModuleNotFoundError: No module named 'src'
    ----------------------------------------
WARNING: Discarding https://files.pythonhosted.org/packages/f6/85/c273089eb6efa5644c0a1382ea553554bc0d40e00a46d989ec67f123f8b5/future-0.18.0.tar.gz#sha256=6142ef79e2416e432931d527452a1cab3aa4a754a0a53d25b2589f79e1106f34 (from https://pypi.org/simple/future/) (requires-python:>=2.6, !=3.0.*, !=3.1.*, !=3.2.*). Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
    ERROR: Command errored out with exit status 1:
     command: 'c:\program files\spyder\python\python.exe' -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\dkoch\\AppData\\Local\\Temp\\pip-install-v2qnl0qq\\future_954b56ac27e44768b990834eb78f8c41\\setup.py'"'"'; __file__='"'"'C:\\Users\\dkoch\\AppData\\Local\\Temp\\pip-install-v2qnl0qq\\future_954b56ac27e44768b990834eb78f8c41\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"') [... cut ...]
      File "C:\Users\dkoch\AppData\Local\Temp\pip-install-v2qnl0qq\canmatrix_c3bb6103bcb146118c22f4c15173d56b\setup.py", line 62, in <module>
        import versioneer
    ModuleNotFoundError: No module named 'versioneer'
    ----------------------------------------
WARNING: Discarding https://files.pythonhosted.org/packages/43/aa/201c1e59d1ce2f9a08cecf138c29f1d0c9bef574439e95ab49240a263275/canmatrix-0.9.tar.gz#sha256=9f1531269880bff3f3bc22a557c1030688e5954ad76d02cbbabc1184af5cdcdf (from https://pypi.org/simple/canmatrix/). Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
```

ebroecker, 2022-03-31:
Is this still an issue? If yes, is this a canmatrix issue?

closed 2022-05-27 by ebroecker