ebroecker/canmatrix #540: TypeError: cannot unpack non-iterable NoneType object

dcetvericov, 2021-01-13:
Hi,

when trying to use ``frame.decode()``, I get the following Exception:
```
...
  File "C:\Users\dmce\AppData\Local\Programs\Python\Python37\lib\site-packages\canmatrix\canmatrix.py", line 2113, in decode
    return self.frame_by_id(frame_id).decode(data)
  File "C:\Users\dmce\AppData\Local\Programs\Python\Python37\lib\site-packages\canmatrix\canmatrix.py", line 1421, in decode
    multiplex_name, multiplex_value = self._get_sub_multiplexer(multiplex_name, multiplex_value, decoded)
TypeError: cannot unpack non-iterable NoneType object
```

The DB I use can be found here: https://www.csselectronics.com/screen/product/obd2-dbc-file
My ArbitrationId object looks like this: canmatrix.ArbitrationId(id=2024, extended=False).
The data is bytearray([0xAA, 0xAA, 0xAA, 0xAA, 0xAA, 0xAA, 0xAA, 0xAA]).

If anything else is needed for debugging, please let me know.

Thank you in advance!

ebroecker, 2021-01-14:
hi @dcetvericov  

I tried to reproduce with current development branch version.
I uses following code without errors

```python
import canmatrix.formats

cm = canmatrix.formats.loadp_flat("/some_path/CSS-Electronics-OBD2-v1.4.dbc")

arbitration_id = canmatrix.ArbitrationId(id=2024, extended=False)
data = bytearray([0xAA, 0xAA, 0xAA, 0xAA, 0xAA, 0xAA, 0xAA, 0xAA])

frame = cm.frame_by_id(arbitration_id)
a = frame.decode(data)
print(a)

# direct decode:
b = cm.decode(arbitration_id, data)
print(b)
```

could you try current development branch version?

dcetvericov, 2021-01-15:
Works like a charm, thank you! Are you planning a new release soon? I would like to use canmatrix as PyPi package for my project.

ebroecker, 2021-01-18:
I created a release just today, could you try?

dcetvericov, 2021-01-19:
Thank you! I will give you feedback once I've tested it.

dcetvericov, 2021-01-21:
Hello @ebroecker,

as promised I've tested your new release today, works like a charm for the case that initially failed. Thank you very much once again :-)

closed 2021-01-21 by dcetvericov