ebroecker/canmatrix #541: seems xlrd does not support xlsx anymore

ebroecker, 2021-01-17:
xlrd.biffh.XLRDError: Excel xlsx file; not supported

```
Traceback (most recent call last):
  File "./test.py", line 100, in <module>
    run_tests()
  File "./test.py", line 85, in run_tests
    canmatrix.convert.convert(in_file, out_file)
  File "/home/travis/build/ebroecker/canmatrix/src/canmatrix/convert.py", line 40, in convert
    dbs = canmatrix.formats.loadp(infile, **options)
  File "/home/travis/build/ebroecker/canmatrix/src/canmatrix/formats/__init__.py", line 77, in loadp
    return load(fileObject, import_type, key, **options)
  File "/home/travis/build/ebroecker/canmatrix/src/canmatrix/formats/__init__.py", line 96, in load
    dbs[key] = module_instance.load(file_object, **options)  # type: ignore
  File "/home/travis/build/ebroecker/canmatrix/src/canmatrix/formats/xlsx.py", line 395, in load
    return xls_loader.load(filename, **options)
  File "/home/travis/build/ebroecker/canmatrix/src/canmatrix/formats/xls.py", line 348, in load
    wb = xlrd.open_workbook(file_contents=file.read())
  File "/home/travis/build/ebroecker/canmatrix/.tox/old_tests/lib/python3.7/site-packages/xlrd/__init__.py", line 170, in open_workbook
    raise XLRDError(FILE_FORMAT_DESCRIPTIONS[file_format]+'; not supported')
xlrd.biffh.XLRDError: Excel xlsx file; not supported
```

wtfch, 2021-05-14:
When I try to convert the xlsx file to DBC file using the converted resault of canconvert. I also got the same error the same with this issue.

AlexDLSy, 2021-11-19:
Having the same issue here.

ebroecker, 2021-11-21:
you could try to use old version of xlrd library xlrd==1.2.0 seemed to work.

Note - this is only a workaround, in future the xlsx-importer have to be written from scratch depending on another library.

AlexDLSy, 2021-11-22:
Would you give some insight on what should be rewritten ? 
Could that also be an opportunity to design the xls(x) format to include more data ?
If a milestone is planned for that, I'd be happy to help, provided that I get explanations about the code structure 😃

ebroecker, 2021-11-22:
hi @AlexDLSy ,

mainly (short term) the "read xlsx"-part hast to be rewritten to use another library.
I have just now startet to rewrite the this part with openpyxl.

long term the xlsx-writing than should also be rewritten to use openpyxl.

Yes, it is possible to store more data in xlsx and reread it again.
Maybe some option for "full" data xslx could be implemented.
Therfor some generic format in excel should be defined.

On the other hand the human readable compact version is popular also.

ebroecker, 2023-07-27:
switched to openpyxl

ebroecker, 2024-06-20:
missing support fpr additionalSignalAttributes and additionalFrameAttributes

closed 2024-06-20 by ebroecker