ebroecker/canmatrix #544: Convert CANdelaStudio CDD or ODX to DBC

bjaschmidt, 2021-01-28:
For our diagnostic datalogging we use ISO15765 UDS service 0x22 ReadDataByIdentifier. These CAN telegrams use same multiplexer as in Vector DBC-files. We describe dataidentifiers with Vector CANdelaStudio in CDD-fileformat and then exports them in ISO22901 Open Diagnostic data eXchange ODX-fileformat.

Both CDD-files and ODX-files defines multiplexers, physical units and scaling in same way as in Vector DBC-files. Note that CDD-files and ODX-files furthermore contains information for flashing and for diagnostic trouble codes, but we do not need conversion of these.

Both CDD-files and ODX-files are XML-fileformats. Do you know of similarities to ARXML-files?

I have tried Erik Moqvist’s “cantools dump” on our CDD-file, but this does not work.
https://github.com/eerimoq/cantools/issues/48

Have you considered making canmatrix read CDD-files or either ODX-files, in order to generate data in DBC-fileformat? 

I thank you for your excellent work

Best regards Bjarne

ebroecker, 2021-01-28:
@bjaschmidt 
Do you have some .cdd example files?
I'll have a look on it

ebroecker, 2021-01-31:
hi @bjaschmidt 

I did some basic implementation in branch iss544_implement_odx_import;

Could you try it?
pip install git+https://github.com/ebroecker/canmatrix.git@iss544

ebroecker, 2022-03-30:
I close this issue here, because basic support is in release already.
If something wrong with it, please create a new issue

closed 2022-03-30 by ebroecker