ebroecker/canmatrix #545: Request: Update the dbc to dbf conversion tool

g-berthiaume, 2021-01-29:
Hi, 

`dbf` file generated by Canmatrix are not identical with the one generated by [Busmaster](https://github.com/rbei-etas/busmaster).
As an example: Busmaster's generated file on the left and Canmatrix's on the right.
[image]

I assume the differences come from Canmatrix using the converter version `1.7.2` instead of the newer `3.2.2`.
To make sure we are compatible with the newer version of Busmaster, would it be possible to update the Canmatrix converter?

Thanks

anphucnguyen, 2021-03-23:
@g-berthiaume Except the differences  in the example picture, it has any deviation between them? If not, I think I can provide the hot fix for your issue 😁

ebroecker, 2021-05-07:
I think this is fixes already

closed 2021-05-07 by ebroecker

g-berthiaume, 2021-05-08:
Great. Thanks for your support.