ebroecker/canmatrix #550: problem while converting arxml to dbc

STX321, 2021-03-04:
C:\Users\Administrator\Desktop\can>canconvert 201115-hd2-EP33L_EP1_FVCM_V02.arxml target2.dbc
INFO - convert - Importing 201115-hd2-EP33L_EP1_FVCM_V02.arxml ...
0
0
0
0
0x7f58
Traceback (most recent call last):
  File "D:\Users\Administrator\AppData\Local\Programs\Python\Python38\Scripts\canconvert-script.py", line 33, in <module>
    sys.exit(load_entry_point('canmatrix==0.9.3', 'console_scripts', 'canconvert')())
  File "D:\Users\Administrator\AppData\Local\Programs\Python\Python38\lib\site-packages\click\core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "D:\Users\Administrator\AppData\Local\Programs\Python\Python38\lib\site-packages\click\core.py", line 782, in main
    rv = self.invoke(ctx)
  File "D:\Users\Administrator\AppData\Local\Programs\Python\Python38\lib\site-packages\click\core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "D:\Users\Administrator\AppData\Local\Programs\Python\Python38\lib\site-packages\click\core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "D:\Users\Administrator\AppData\Local\Programs\Python\Python38\lib\site-packages\canmatrix\cli\convert.py", line 135, in cli_convert
    canmatrix.convert.convert(infile, outfile, **options)
  File "D:\Users\Administrator\AppData\Local\Programs\Python\Python38\lib\site-packages\canmatrix\convert.py", line 40, in convert
    dbs = canmatrix.formats.loadp(infile, **options)
  File "D:\Users\Administrator\AppData\Local\Programs\Python\Python38\lib\site-packages\canmatrix\formats\__init__.py", line 77, in loadp
    return load(fileObject, import_type, key, **options)
  File "D:\Users\Administrator\AppData\Local\Programs\Python\Python38\lib\site-packages\canmatrix\formats\__init__.py", line 94, in load
    dbs = module_instance.load(file_object, **options)  # type: ignore
  File "D:\Users\Administrator\AppData\Local\Programs\Python\Python38\lib\site-packages\canmatrix\formats\arxml.py", line 1948, in load
    result.update(decode_can_helper(ea, float_factory, ignore_cluster_info))
  File "D:\Users\Administrator\AppData\Local\Programs\Python\Python38\lib\site-packages\canmatrix\formats\arxml.py", line 1843, in decode_can_helper
    frame = get_frame(frameTrig, ea, multiplex_translation, float_factory)
  File "D:\Users\Administrator\AppData\Local\Programs\Python\Python38\lib\site-packages\canmatrix\formats\arxml.py", line 1507, in get_frame
    get_frame_from_container_ipdu(pdu, new_frame, ea, float_factory)
  File "D:\Users\Administrator\AppData\Local\Programs\Python\Python38\lib\site-packages\canmatrix\formats\arxml.py", line 1364, in get_frame_from_container_ipdu
    get_signals(pdu_sig_mapping, target_frame, ea, header_id, float_factory, bit_offset=header_length+offset)
  File "D:\Users\Administrator\AppData\Local\Programs\Python\Python38\lib\site-packages\canmatrix\formats\arxml.py", line 1235, in get_signals
    new_signal.initial_value = float_factory(initvalue.text)
decimal.InvalidOperation: [<class 'decimal.ConversionSyntax'>]

Hi, I am using canmatrix for converting arxml to dbc.
Please let me know how I can solve it

Best Regards
STX

ebroecker, 2021-03-04:
could you please try branch `iss550_arxml_workarounds` ?

pip install git+https://github.com/ebroecker/canmatrix.git@iss550_arxml_workarounds

You should get a log-output for exact this code line which failed

ebroecker, 2021-05-07:
seems to work now?

closed 2021-05-07 by ebroecker

cone95, 2021-09-29:
Hi! I was having the same problem and the solution worked for me. However, the branch iss550_arxml_workarounds seems no longer to be present.

ebroecker, 2021-09-29:
Hi  @cone95 

this fix should be integrated in develope branch already