ebroecker/canmatrix #558: Can't convert to ArXML

Dan13l-N, 2021-03-31:
Hello, I have some DBC files I'm trying to convert to ARXML, but every file I try I get:

C:\Python\Scripts>canconvert -v B8.dbc b8.arxml
arxml is not supported
kcd is not supported
fibex is not supported
xls is not supported
xlsx is not supported
yaml is not supported
INFO - convert - Importing B8.dbc ...
INFO - convert - done

INFO - convert - Exporting b8.arxml ...
INFO - convert -
INFO - convert - 124 Frames found
ERROR - __init__ - This file format is not supported for writing
INFO - convert - done

And no files are created. However, from docs and code I see writing to ArXML is supported?

Thanks a lot, Daniel

anphucnguyen, 2021-03-31:
Hi Daniel,

>arxml is not supported
kcd is not supported
fibex is not supported
xls is not supported
xlsx is not supported
yaml is not supported

Let's install the dependency first by command: pip install -r requirements.txt

As I remember that the canconvert does not support dbc -> arxml,  it support arxml ->dbc.

ebroecker, 2021-04-02:
Hi    @Dan13l-N,

yes @anphucnguyen  is right, you need the dependencies at least you need lxml.

But canmatrix does support converting to ARXML.

Best regards

Dan13l-N, 2021-04-08:
Thanks a lot! I confirm it works now!! Great!

closed 2021-04-08 by Dan13l-N