ebroecker/canmatrix #565: initial value issue

jazi007, 2021-04-11:
Hello,

Using canmatrix for parsing arxml file I faced an crash due to initial_value problem.
by default initial value is set to 0 but for some signal it's out of range.

here's an example of valid signal: s = {min=69, max=100.5, size=6, offset=69, factor=0.5}

https://github.com/ebroecker/canmatrix/blob/development/src/canmatrix/canmatrix.py#L584 **(pack_bitstring)**
this function is going to return:

```python
>>> '{:0{}b}'.format(int((2<<6)+ ((0 - 69) / 0.5)), 6)[-6:]
'-01010'
```

I did a quick fix: https://github.com/jazi007/canmatrix/commit/0573389f9b11578858261cad64265270fe2f7127

ebroecker, 2021-04-12:
Hi @jazi007 ,

could you create a PR?

jazi007, 2021-04-12:
Hello @ebroecker,

Done : https://github.com/ebroecker/canmatrix/pull/566

ebroecker, 2021-04-15:
merged

closed 2021-04-15 by ebroecker