ebroecker/canmatrix #568: canconvert command in command line returns arxml is not support. I'm unable to convert the arxml file to dbc format.

jatinpatel1489, 2021-04-29:


jatinpatel1489, 2021-04-29:
After downloading the lxml from Github resolved the arxml is not supported error. However, it doesn't create a DBC file. Please see the log messages below. 

xls is not supported
xlsx is not supported
yaml is not supported
INFO - convert - Importing C:\Users\PatelJV\OneDrive - Jacobs\GMAL\neoVI Plugin (LIN)\Network Databases\Sample.arxml ...
DEBUG - arxml - Read arxml ...
DEBUG - arxml - Build arTree ...
DEBUG - arxml - 0 frames in arxml...
DEBUG - arxml - 0 can-frame-triggering in arxml...
DEBUG - arxml - 0 SIGNAL-TO-PDU-MAPPINGS in arxml...
DEBUG - arxml - 4 I-SIGNAL-TO-I-PDU-MAPPING in arxml...
INFO - convert - done

INFO - convert - Exporting C:\Users\PatelJV\Sample.dbc ...
INFO - convert - done

Please advise if there is something is missing. Thanks!

ebroecker, 2021-04-30:
Hi @jatinpatel1489 

what arxml is it?
DEBUG - arxml - **0** frames in arxml...
DEBUG - arxml - **0** can-frame-triggering in arxml...

you could try:
_--arxmlIgnoreClusterInfo_

jatinpatel1489, 2021-04-30:
The arxml file I was using was just a sample file without much information. Your suggestion helped to ignore the cluster information and I could see the DBC file now. Thanks a bunch.

closed 2021-04-30 by jatinpatel1489