ebroecker/canmatrix #572: Enum representation in dbc files not correct when converting from sym file

MichaelBuchner, 2021-05-07:
If an enum is stored in the sym file for a variable, then this enum is not correctly created in the dbc file. The link of the enum table with the enum variable is missing. Enclosed is a brief example:

This is the sym file:

```
FormatVersion=5.0 // Do not edit this line!
Title="Test-CAN"

{ENUMS}
enum Status_val(0="R", 1="B", 2="E")

{SENDRECEIVE}

[Flow_extCAN]
ID=700h
DLC=6
Var=status Status_val 36,2 
```

This is how the dbc file should look:
```
VERSION ""

NS_ :
  CM_
  BA_DEF_
  BA_
  VAL_
  BA_DEF_DEF_
  SIG_VALTYPE_

BS_:

BU_: Vector__XXX

BO_ 1792 Flow_extCAN: 6 Vector__XXX
SG_ status : 36|2@1+ (1,0) [0|3] "" Vector__XXX

BA_DEF_  "BusType" STRING ;

BA_DEF_DEF_  "BusType" "CAN";

BA_ "BusType" "CAN";
VAL_ 1792 status 0 "R" 1 "B" 2 "E" ; 
```

This is how it is looking:
```
VERSION ""

NS_ :

BS_:

BU_: Vector__XXX
VAL_TABLE_ Status_val 0 "R" 1 "B" 2 "E";

BO_ 1792 Flow_extCAN: 6 Plutron
 SG_ status : 36|2@1+ (1,0) [0|3] "" Vector__XXX

BA_DEF_  "BusType" STRING ;

BA_DEF_DEF_ "BusType" "CAN";

BA_ "BusType" "CAN";

BA_DEF_  "Title" STRING;
BA_DEF_DEF_ "Title" "canmatrix-Export";

BA_ "Title"   "Test-CAN"; 
```

ebroecker, 2021-05-07:
Hi @MichaelBuchner ,

Thanks for reporting this bug!

could you try branch  `iss572_sym_valtab` ?

MichaelBuchner, 2021-05-25:
Hi @ebroecker,
we tryed the fix and so fare it seems working. The only thing that I do not understand is the VALUE_TABLE line. For me this line seems to be not needed. If you open and save the dbc file with an (very old :-) ) dbc file editor from vector this lines are removed from the dbc file.

MichaelBuchner, 2021-05-25:
Hi @ebroecker,
I tested the dbc files now with a new version of the Vector DBC-File editor. Here the VALUE_TABLE lines are not removed. (but still not used for something it seems) Anyway it seems sorking :-)

commit 71d03792 (2021-08-18): fixes #572
closed 2021-08-18 by ebroecker