ebroecker/canmatrix #573: DBC STRING type attribute includes trailing whitespace from regex

yesralph, 2021-05-11:
In `formats/dbc.py`, a common `regexp` pattern ends with `+(.+) *; *`, which includes the last matching group with trailing whitespace if there are whitespace before `;`. A simplified example is:
```
regexp = re.compile('^BA_ +(.+) *; *')
ba = regexp.match('BA_ "TEXT" ;')
ba.group(1)     # '"TEXT" '
```

The issue shows up when we remove the double quotes `"`, starting around `line 947`, eg.:
```
signal.attributes[define] = signal.attributes[define][1:-1]
```
And `ba.group(1)` above becomes `TEXT"`.

I think either we should make the pattern `+(\S+) *; *` to avoid extra whitespace matching into the last group, or we should `strip()` the last matching group before it is assigned to attribute.

yesralph, 2021-05-11:
Actually, `Frame.add_attribute()` handles the attribute type properly. Can we update other `add_attribute()` as well?
```
Class Frame:
    ...
    def add_attribute(self, attribute, value):
        # type: (str, typing.Any) -> None
        """
        Add the attribute with value to customer Frame attribute-list. If Attribute already exits, modify its value.
        :param str attribute: Attribute name
        :param any value: attribute value
        """
        try:
            self.attributes[attribute] = str(value)
        except UnicodeDecodeError:
            self.attributes[attribute] = value
        if type(self.attributes[attribute]) == str:
            self.attributes[attribute] = self.attributes[attribute].strip()

```

ebroecker, 2022-03-31:
is this issue still open?

closed 2022-05-27 by ebroecker