ebroecker/canmatrix #580: Convert arxml to DBC  TypeError: exceptions must be old-style classes or derived from BaseException, not str

orangewsc, 2021-06-02:
Hi,when we use the tool convert arxml to DBC, input file: file1.arxml and file2.arxml.
 convert file1.arxml: OK
but convert file2.arxml: TypeError: exceptions must be old-style classes or derived from BaseException, not str;
we try python2.7 and python3.7, the error: exceptions must derive from BaseException still exists.

Finaly,I want to know, when convert DBC to arxml, whether the output arxml schema version is only 3.2.3?

Thanks

ebroecker, 2021-07-08:
some 4.X schema is supported as well
`--arxmlExportVersion`
https://canmatrix.readthedocs.io/en/latest/cli.html#commandline-overview

gaurav6996, 2021-07-12:
Hi dev,
I am also facing a similar issue,
ipdu = ea.get_child(static_part, "I-PDU")
  File "...anaconda3\lib\site-packages\canmatrix\formats\arxml.py", line 215, in get_child
    raise "use follow_ref!"
TypeError: exceptions must derive from BaseException

Do we have a solution here? Or is the issue with the ARXML file itself?

Thank you.

ebroecker, 2021-07-12:
@gaurav6996 
this is an Error in arxml-module.
thanks for reporting.

ebroecker, 2022-03-31:
should be fixed

closed 2022-03-31 by ebroecker