ebroecker/canmatrix #581: Default value for unset signals or unused bits in the frame should be '1' for J1939

linzbdav, 2021-06-23:
The default value for unset signals or unused bits in the frame is currently set to '0' but the document SAE J1939-71 in Table 1 defines if a signal is not available or nor requested the value should be 0xFF. Therefore the unset signals ans unused bits should be set to '1'.

ebroecker, 2023-01-23:
hi @linzbdav 

I was just looking to this issue...

Not sure how to fix this *compatible*, because I *think* default value should be 0 for *normal can* and seems it should be 1 (0xff) for j1939.

What use case you came along with this bug (converting, creation, what ever)?
I just want to know this to find a useful workaround ...

linzbdav, 2023-04-11:
Hi @ebroecker 
Sorry for my late response I was very busy during the last days. I use the library for creation of CAN-frames, I use a dbc file with all message definition and want to generate the frame with, some messages don't use all bits or I don't want to set a signal then the resulting value in the message should be filled with ones.

ebroecker, 2023-07-18:
Hi @linzbdav 

do you use canmatrix to create the frame? With èncode`?

closed 2026-10-16 by ebroecker