ebroecker/canmatrix #582: convert from .xls to .dbc sets all DLC's to 8

luisgongod, 2021-06-24:
After following the step for additional formats 
`$ pip install git+https://github.com/ebroecker/canmatrix#egg=canmatrix[kcd]`

Converting from .dbc to .xls is fine. But converting back to .dbc makes all DLC to 8. 

original dbc:
`BO_ 2180521234 SENSOR: 4 SENSOR
`
.xls to .dbc
`BO_ 2180521234 SENSOR: 8 SENSOR
`
I "solve" this by changing : `dlc = 8` to `dlc=0` in method `load` in file `xls.py` line 422

I dont know if this is the right fix. Maybe would be good to include a DLC column in the xls file

ebroecker, 2021-07-06:
maybe something like
`--additionalFrameAttributes=length`

helps

luisgongod, 2021-10-06:
Hi, sorry for the late reply. The project was idle for a while.

I can confirm `--additionalFrameAttributes=size` Solves the "issue".  How I tested:
Converting the `.dbc` with the proper DLC values to `xls`
```
canconvert matrix.dbc matrix.xls --additionalFrameAttributes=size
```
and back to dbc
```
canconvert matrix.xls matrix_2.dbc
```
also tried that any modifications of the _frame.size_ column in the xls file will change the dbc's DLC

we can close this :)