ebroecker/canmatrix #584: AttributeError: module 'canmatrix' has no attribute 'CanMatrix'

merna-sedki, 2021-07-08:
i get this error when i try to convert arxml to dbc my code is as follows:
from canmatrix import convert
convert.convert("in.arxml", "out.dbc")

here is error-log:
xls is not supported
xlsx is not supported
Traceback (most recent call last):
  File "C:/workset/castleWorkset/proj3480_utilities/HMIs/A4_HMI/test.py", line 2, in <module>
    from canmatrix.cli import convert
  File "C:\Python\lib\site-packages\canmatrix\__init__.py", line 11, in <module>
    from canmatrix.canmatrix import (
  File "C:\Python\lib\site-packages\canmatrix\canmatrix.py", line 42, in <module>
    from future.moves.itertools import zip_longest
  File "C:\Python\lib\site-packages\future\moves\__init__.py", line 8, in <module>
    import_top_level_modules()
  File "C:\Python\lib\site-packages\future\standard_library\__init__.py", line 810, in import_top_level_modules
    with exclude_local_folder_imports(*TOP_LEVEL_MODULES):
  File "C:\Python\lib\site-packages\future\standard_library\__init__.py", line 781, in __enter__
    module = __import__(m, level=0)
  File "C:\workset\castleWorkset\proj3480_utilities\HMIs\A4_HMI\test.py", line 10, in <module>
    convert.convert("E3_1_2_Premium_V12.04.25F_P21_AR430_20210320_LSG_R_BP_fixed.arxml", "file.dbc")
  File "C:\Python\lib\site-packages\canmatrix\convert.py", line 40, in convert
    dbs = canmatrix.formats.loadp(infile, **options)
  File "C:\Python\lib\site-packages\canmatrix\formats\__init__.py", line 77, in loadp
    return load(fileObject, import_type, key, **options)
  File "C:\Python\lib\site-packages\canmatrix\formats\__init__.py", line 94, in load
    dbs = module_instance.load(file_object, **options)  # type: ignore
  File "C:\Python\lib\site-packages\canmatrix\formats\arxml.py", line 1950, in load
    result.update(decode_can_helper(ea, float_factory, ignore_cluster_info))
  File "C:\Python\lib\site-packages\canmatrix\formats\arxml.py", line 1791, in decode_can_helper
    db = canmatrix.CanMatrix()
AttributeError: module 'canmatrix' has no attribute 'CanMatrix'

merna-sedki, 2021-07-08:
@ebroecker am i doing something wrong ? thanks in advance for ur  support

ebroecker, 2021-07-08:
no, it should work.
But why don't you use something like
`python -m canmatrix.cli.convert in.arxml out.dbc`

Did you install the package (something like `pip install canmatrix`)?

closed 2022-03-31 by ebroecker