ebroecker/canmatrix #595: pdus=[] when using canmatrix.formats.loadp to get information in arxml files

smzhang102, 2021-08-05:
hey,

I tried to extract frame information in arxml using canmatrix.formats.loadp, and met an issue that the pdus under frames is []. meanwhile, some pdu names were also missing in the signalGroups. But I can find the PDU names in PDU-Mapping in I-SIGNAL-I-PDU in autosar. 

By the way, the periods of messages are  in pdu names, e.g. BMS_100ms_PDU60, not always. I find some periods in the description of signals, e.g. "period(ms):100", could canmatrix extract the description?

Maybe you can give me some hints on how to get it?

The canmatrix I used was downloaded from Github, which I thought is the latest, right?

ebroecker, 2022-03-31:
is this issue still open?

closed 2022-05-27 by ebroecker

pempem98, 2023-12-29:
It still happens.
Maybe the issue starts from [here](https://github.com/ebroecker/canmatrix/blob/adcd8955d162d4119b8359bccd542a5db8baa3a6/src/canmatrix/formats/arxml.py#L1554C18-L1554C18)