ebroecker/canmatrix #596: Error During loading of DBC (canmatrix.formats.loadp(self.path)

PUNITPNT, 2021-08-14:
hi,
My Requirement:
I am converting DBC to Excel, according to candbc++ editor excel format using python 3.9.
cantools latest version.

Problem 1 
  ERROR 1:
  Error during loading of DBC (canmatrix.formats.loadp(self.path_of_dbc))
  
  error with line no: 67
  b' SG_ A_B_C_D : 61|2@0+ (1,0) [0|3] "-\r\n'     (A_B_C_D i replaced with actual signal name, as i cant share actual signal name)
  
  error with line no: 576
  b' SG_ A_B_C_D_E : 39|16@0+ (0.01,0) [0|655.35] "km/h\r\n'

   error with line no: 582
  b'BA_ "DescriptionJ" SG_ 123 A_B_C_D_E_F "";\r\n'
  
  error with line no: 2517
  b'BA_ "GenSigSendType" SG_ 123 A_B_C_D_E_F 0;\r\n'

Problem 2:
big endian start bit is different from as it present in vector Candbc++ editor.

Problem 3:
i am getting initial value from cantools as NONE , but initial value present in dbc and visible in  vector Candbc++ editor.

Note: I not able to share you the dbc file.

I am new to git hub.

ebroecker, 2021-08-16:
so - your dbc is not clean as it is. 
A signal-definition line should not and as:

`SG_ A_B_C_D_E : 39|16@0+ (0.01,0) [0|655.35] "km/h\r\n"`

but usually there should be at least one receiver ECU like:

`SG_ A_B_C_D_E : 39|16@0+ (0.01,0) [0|655.35] "km/h\r\n" SOME_RECEIVER_ECU`

do you have other tools, that can work with this file?

I am able to provide a workaround also - please try branch https://github.com/ebroecker/canmatrix/tree/iss_596

`pip install git+https://github.com/ebroecker/canmatrix.git@iss_596`

PUNITPNT, 2021-08-16:
hi 

@ebroecker 
I can open this kind of dbc in Vector Candbc++ editor tool.

ebroecker, 2021-08-16:
@PUNITPNT 
does it change, if you save it again?

PUNITPNT, 2021-08-16:
hi @ebroecker 

when i opened dbc in Vector Candbc++ editor tool. I able to see that some of the signals are not defined with any receiver attribute.

Right Configuration : i think every signal should have receiver , as you said it should have at least one receiver

but in notepad it is like SG_ xxxx : 6|3@0+ (1,0) [0|7] "" Vector__XXX

branch  iss_596 not able to install because of some error, i think it require git

ERROR: Cannot find command 'git' - do you have 'git' installed and in your PATH?