ebroecker/canmatrix #607: Canconvert-File not generated

Yuvalekha-se, 2021-11-10:
C:\Users\Yuvalekha RaviKumar\canmatrix-development\src\canmatrix\cli>python convert.py --arxmlExportVersion 4.1.0 C:\Users\canmatrix-development\can\base.arxml C:\Users\canmatrix-development\can\target.dbc
INFO - convert - Importing C:\Users\canmatrix-development\can\base.arxml ...
INFO - convert - done

INFO - convert - Exporting C:\Users\canmatrix-development\can\target.dbc ...
INFO - convert - done

--------------------------------------------
There was no errors on executing the canconvert 
but no files have created

It would be nice if you are helping me to solve this

ebroecker, 2021-11-12:
`--arxmlExportVersion 4.1.0 `
does make no sense, because you import an arxml.

is there any can-bus defined in "base.arxml"?

Did you try `--arxmlIgnoreClusterInfo`

ebroecker, 2022-03-31:
does this help?

closed 2022-05-27 by ebroecker