ebroecker/canmatrix #614: DBC --> XLS then XLS --> DBC does not preserve values

AlexDLSy, 2021-11-19:
I used canconvert on my dbc file to create an xls file, then used canconvert on the created xls file to output another dbc file.

I expected both dbc files to contain the same values, but the resulting dbc not only has had many deleted attributes, some values have been changed. The list of differences using cancompare is huge.

Is that intended ? Where can I find the coverage of conversion between xls and dbc ? (Ie what is exported in xls from dbc and what is not, and in the reverse direction as well)
Currently, the conversion from xlsx is not possible due to #541 but it might have the same issue.

ebroecker, 2021-11-21:
Hi @AlexDLSy ,

yes, xls(x) is a lossy format.
It was not designed for the workflow you use it.

You can make things better by exporting some extra attributes to xls(x) with "–additionalSignalAttributes and –additionalFrameAttributes"

Some signal-attributes which will help:
- min
- max
- is_signed
- factor
- offset

wienans, 2022-03-15:
Hi @ebroecker 
i used the additionalSignalAttributes flag
canconvert --additionalSignalAttributes min,max,is_signed,factor,offset can_dbc.dbc test1.xls
and later tried to converted it back with
canconvert --additionalSignalAttributes min,max,is_signed,factor,offset test1.xls can_dbc_test.dbc

But i incountered an error:
`Traceback (most recent call last):
  File "/home/sven/.local/bin/canconvert", line 33, in <module>
    sys.exit(load_entry_point('canmatrix==0.9.4.post2+gf56652f', 'console_scripts', 'canconvert')())
  File "/home/sven/.local/lib/python3.6/site-packages/click/core.py", line 1128, in __call__
    return self.main(*args, **kwargs)
  File "/home/sven/.local/lib/python3.6/site-packages/click/core.py", line 1053, in main
    rv = self.invoke(ctx)
  File "/home/sven/.local/lib/python3.6/site-packages/click/core.py", line 1395, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/sven/.local/lib/python3.6/site-packages/click/core.py", line 754, in invoke
    return __callback(*args, **kwargs)
  File "/home/sven/.local/lib/python3.6/site-packages/canmatrix/cli/convert.py", line 153, in cli_convert
    canmatrix.convert.convert(infile, outfile, **options)
  File "/home/sven/.local/lib/python3.6/site-packages/canmatrix/convert.py", line 331, in convert
    canmatrix.formats.dumpp(out_dbs, out_file_name, **options)
  File "/home/sven/.local/lib/python3.6/site-packages/canmatrix/formats/__init__.py", line 129, in dumpp
    dump(db, file_object, export_type, **options)
  File "/home/sven/.local/lib/python3.6/site-packages/canmatrix/formats/__init__.py", line 104, in dump
    module_instance.dump(can_matrix_or_cluster, file_object, **options)  # type: ignore
  File "/home/sven/.local/lib/python3.6/site-packages/canmatrix/formats/dbc.py", line 236, in dump
    if signal.phys2raw(None) != 0:
  File "/home/sven/.local/lib/python3.6/site-packages/canmatrix/canmatrix.py", line 432, in phys2raw
    raw_value = (value - self.offset) / self.factor
TypeError: unsupported operand type(s) for -: 'decimal.Decimal' and 'float'
`
It looks like it only appears if i the factor and offset are exported to the xls file

Karstensson, 2022-07-13:
I have a fix for this, but don't seem to have access to upload a branch for this, therefore I cannot create a PR.
But line 432 in canmatrix.py needs to be replaced by

`raw_value = (value - self.offset) / self.factor`

to 
`raw_value = (value - decimal.Decimal(self.offset)) / self.factor`

ebroecker, 2022-07-13:
@Karstensson 
thanks, I'll doublecheck and include your fix.

closed 2023-06-22 by ebroecker