ebroecker/canmatrix #625: dbc attribute string exported with wrong value

AnNeub, 2022-02-14:
If the parameter 'value' is 0 (zero) than the elif condition gets true and the value will be exported as "", which breaks the dbc file if the attribute is defined as INT.
If 'value' is None maybe the attribute_string shall not be exported at all, so that the default value is taken.

```
def create_attribute_string(attribute, attribute_class, name, value, is_string):
    # type: (str, str, str, typing.Any, bool) -> str
    if is_string:
        value = '"' + value + '"'
    elif not value:  # <-- Bug
        value = '""'

    attribute_string = 'BA_ "' + attribute + '" ' + attribute_class + ' ' + name + ' ' + str(value) + ';\n'
    return attribute_string
```

ebroecker, 2022-02-15:
Hi @AnNeub ,

I tried to reproduce that error but was not successful.

Could you provide test-data where this code causes an error?

Eduard

AnNeub, 2022-02-16:
Hi @ebroecker 

here some short example which leads to the same error.

BR Andreas

```
import canmatrix
import canmatrix.formats

def main():
    cm = canmatrix.CanMatrix()
    cm.add_ecu_defines('NmStationAddress', 'INT 0 255')
    ecu = canmatrix.Ecu('TestEcu')
    ecu.add_attribute('NmStationAddress', 0)
    cm.add_ecu(ecu)
    frame = canmatrix.Frame(name='name',
                            arbitration_id=canmatrix.ArbitrationId(0x123,False),
                            size=8)
    cm.add_frame(frame)
    canmatrix.formats.dumpp({'': cm}, 'Test.dbc')

if __name__ == '__main__':
    main()
```

```
VERSION "created by canmatrix"

NS_ :

BS_:

BU_: TestEcu 

BO_ 291 name: 8 Vector__XXX

BA_DEF_ BU_ "NmStationAddress" INT 0 255;
BA_ "NmStationAddress" BU_ TestEcu "";
```

ebroecker, 2022-02-16:
Hi AnNeub,

thanks for providing this example.

Funfact: This problem did not occur for frame-attributes this is why I did not reproduce it.

You could try this branch https://github.com/ebroecker/canmatrix/tree/iss_625 where it should be fixed

BR Eduard

AnNeub, 2022-03-22:
Hi @ebroecker 

Sorry for the delay, 
fix looks good.

BR Andreas

closed 2022-03-22 by AnNeub