ebroecker/canmatrix #632: Several errors when converting ARXML file to DBC file

kw8088, 2022-03-24:
I face following problems when trying to convert an Autosar 3.2.2 ARXML file to DBC file 
Please help

ldf is not supported
xls is not supported
INFO - convert - Importing VS30_HMI2_CAN_2021_42b_2023.arxml ...
ERROR - arxml - No valid compu method found for isignal/systemsignal <somesignals> check ARXML file!!
ERROR - arxml - No valid compu method found for isignal/systemsignal <somesignals> check ARXML file!!
ERROR - arxml - No valid compu method found for isignal/systemsignal <somesignals> check ARXML file!!
ERROR - arxml - No valid compu method found for isignal/systemsignal <somesignals> check ARXML file!!
ERROR - arxml - No valid compu method found for isignal/systemsignal <somesignals> check ARXML file!!
Traceback (most recent call last):
  File "/home/user/.local/bin/canconvert", line 8, in <module>
    sys.exit(cli_convert())
  File "/home/user/.local/lib/python3.5/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/home/user/.local/lib/python3.5/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/home/user/.local/lib/python3.5/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/user/.local/lib/python3.5/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/home/user/.local/lib/python3.5/site-packages/canmatrix/cli/convert.py", line 153, in cli_convert
    canmatrix.convert.convert(infile, outfile, **options)
  File "/home/user/.local/lib/python3.5/site-packages/canmatrix/convert.py", line 68, in convert
    dbs = canmatrix.formats.loadp(infile, **options)
  File "/home/user/.local/lib/python3.5/site-packages/canmatrix/formats/__init__.py", line 71, in loadp
    return load(fileObject, import_type, key, **options)
  File "/home/user/.local/lib/python3.5/site-packages/canmatrix/formats/__init__.py", line 88, in load
    dbs = module_instance.load(file_object, **options)  # type: ignore
  File "/home/user/.local/lib/python3.5/site-packages/canmatrix/formats/arxml.py", line 1959, in load
    result.update(decode_can_helper(ea, float_factory, ignore_cluster_info))
  File "/home/user/.local/lib/python3.5/site-packages/canmatrix/formats/arxml.py", line 1852, in decode_can_helper
    frame = get_frame(frameTrig, ea, multiplex_translation, float_factory, headers_are_littleendian)
  File "/home/user/.local/lib/python3.5/site-packages/canmatrix/formats/arxml.py", line 1595, in get_frame
    get_signals(pdu_sig_mapping, new_frame, ea, None, float_factory)
  File "/home/user/.local/lib/python3.5/site-packages/canmatrix/formats/arxml.py", line 1135, in get_signals
    get_sys_signals(system_signal, system_signals, frame, group_id, ea)
NameError: name 'get_sys_signals' is not defined

ebroecker, 2022-03-24:
hi @kw8088 ,

thanks for reporting this bug.

Could you please try branch `iss_632`?

`pip install git+https://github.com/ebroecker/canmatrix.git@iss_632`

At least the Error should be resolved

kw8088, 2022-03-25:
Hi @ebroecker ,
It works now! Thanks a lot!

closed 2022-03-30 by ebroecker