ebroecker/canmatrix #634: canconvert -h gives errors

trvini, 2022-03-27:
Hi,

I installed canmatrix on ubuntu machine. 
i also tried installing the branch iss_632 as mentioned in the issue [https://github.com/ebroecker/canmatrix/issues/632](url) , 
but i still get the following error.
Thanks for your support.

```
bm@up2-2:~$ cat /etc/os-release 
NAME="Ubuntu"
VERSION="20.04.4 LTS (Focal Fossa)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 20.04.4 LTS"
VERSION_ID="20.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=focal
UBUNTU_CODENAME=focal
bm@up2-2:~$ python --version
Python 2.7.18
bm@up2-2:~$ python3 --version
Python 3.8.10
bm@up2-2:~$ canconvert -h
ldf is not supported
Traceback (most recent call last):
  File "/home/bm/.local/bin/canconvert", line 5, in <module>
    from canmatrix.cli.convert import cli_convert
  File "/home/bm/.local/lib/python3.8/site-packages/canmatrix/__init__.py", line 8, in <module>
    import canmatrix.formats as formats
  File "/home/bm/.local/lib/python3.8/site-packages/canmatrix/formats/__init__.py", line 25, in <module>
    importlib.import_module("canmatrix.formats." + module)
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "/home/bm/.local/lib/python3.8/site-packages/canmatrix/formats/xls.py", line 35, in <module>
    import xlwt
  File "/home/bm/.local/lib/python3.8/site-packages/xlwt/__init__.py", line 5, in <module>
    from .Row import Row
  File "/home/bm/.local/lib/python3.8/site-packages/xlwt/Row.py", line 9, in <module>
    from . import ExcelFormula
  File "/home/bm/.local/lib/python3.8/site-packages/xlwt/ExcelFormula.py", line 8, in <module>
    from . import ExcelFormulaParser, ExcelFormulaLexer
  File "/home/bm/.local/lib/python3.8/site-packages/xlwt/ExcelFormulaLexer.py", line 53, in <module>
    _re = recompile(
  File "/usr/lib/python3.8/re.py", line 252, in compile
    return _compile(pattern, flags)
  File "/usr/lib/python3.8/re.py", line 304, in _compile
    p = sre_compile.compile(pattern, flags)
  File "/usr/lib/python3.8/sre_compile.py", line 764, in compile
    p = sre_parse.parse(p, flags)
  File "/usr/lib/python3.8/sre_parse.py", line 958, in parse
    p.state.flags = fix_flags(str, p.state.flags)
  File "/usr/lib/python3.8/sre_parse.py", line 925, in fix_flags
    raise ValueError("cannot use LOCALE flag with a str pattern")
ValueError: cannot use LOCALE flag with a str pattern
bm@up2-2:~$ 
```

ebroecker, 2022-03-29:
... seems to be an issue of some specific xlwt version,
which xlwt-version do you have installed?

trvini, 2022-03-29:
Thankyou for your reply.
 I installed the latest version as mentioned in your install instructions' from https://pypi.org/project/xlwt-future/ .

ebroecker, 2022-03-29:
could you try xlwt-version 1.3.0?

closed 2022-04-05 by ebroecker