ebroecker/canmatrix #635: xlsx to dbc something wrong

Ederwardlu, 2022-03-29:
who know how to fix the erro
[image]
r

Ederwardlu, 2022-03-29:
> you could try to use an old xlrd-version:
> `pip install xlrd==1.2.0`

my xldd version is 1.2.0

ebroecker, 2022-03-29:
hi @Ederwardlu ,

ok, hmm I cannot reproduce for now.

Can you create some dummy-sample file where you get that error?

Ederwardlu, 2022-03-30:
> hi @Ederwardlu ,
> 
> ok, hmm I cannot reproduce for now.
> 
> Can you create some dummy-sample file where you get that error?

The follow is my python lib
[image]

[image]

[image]

when I try to convert dbc to xlsx, as follo
[image]

[image]
ws
[image]
then I modify the "hh2.xlsx",something wrong 
[image]

ebroecker, 2022-03-30:
Hi @Ederwardlu,

I am working on changing the used excel-libraries (switching to openpyxl).
You could give it a try in branch `iss541`
`pip install git+https://github.com/ebroecker/canmatrix.git@iss541`

Not sure, that this helps in your issue here

Ederwardlu, 2022-04-11:
> Hi @Ederwardlu,
> 
> I am working on changing the used excel-libraries (switching to openpyxl). You could give it a try in branch `iss541` `pip install git+https://github.com/ebroecker/canmatrix.git@iss541`
> 
> Not sure, that this helps in your issue here

hi, I have troubles on how to convert xlsx to J1939 doc,now I can convert xlsx to doc,but the template of doc is canned

Ederwardlu, 2022-04-11:
> > Hi @Ederwardlu,
> > I am working on changing the used excel-libraries (switching to openpyxl). You could give it a try in branch `iss541` `pip install git+https://github.com/ebroecker/canmatrix.git@iss541`
> > Not sure, that this helps in your issue here
> 
> hi, I have troubles on how to convert xlsx to J1939 doc,now I can convert xlsx to doc,but the template of doc is cannfd

closed 2022-04-11 by Ederwardlu