ebroecker/canmatrix #640: Problem with convertion arxml to dbc

dragoudin, 2022-05-06:
Whein I convert my arxml file, I get the following error:
```
  File "C:\Python\Python310\lib\site-packages\canmatrix\formats\__init__.py", line 88, in load
    dbs = module_instance.load(file_object, **options)  # type: ignore
  File "C:\Python\Python310\lib\site-packages\canmatrix\formats\arxml.py", line 1937, in load
    ea.open(file)
  File "C:\Python\Python310\lib\site-packages\canmatrix\formats\arxml.py", line 84, in open
    self.fill_caches()
  File "C:\Python\Python310\lib\site-packages\canmatrix\formats\arxml.py", line 71, in fill_caches
    new_ar_path = self.fill_caches(sub_element, ar_path)
  File "C:\Python\Python310\lib\site-packages\canmatrix\formats\arxml.py", line 71, in fill_caches
    new_ar_path = self.fill_caches(sub_element, ar_path)
  File "C:\Python\Python310\lib\site-packages\canmatrix\formats\arxml.py", line 71, in fill_caches
    new_ar_path = self.fill_caches(sub_element, ar_path)
  [Previous line repeated 5 more times]
  File "C:\Python\Python310\lib\site-packages\canmatrix\formats\arxml.py", line 73, in fill_caches
    ar_path += '/' + new_ar_path
TypeError: can only concatenate str (not "NoneType") to str
```
Seems a field in arxml is empty and it cause an error

ALUE, 2022-05-06:
您的邮件已收到！              祝好             刘毅

ebroecker, 2022-12-15:
anything new here?
Maybe your arxml is corrupt?
Some example-file?
Maybe some empty SHORT-NAME-Tag?

closed 2023-01-24 by ebroecker