ebroecker/canmatrix #643: ldf is not supported

hongzhifu, 2022-06-22:
I try to convert file from arxml  to dbc ,then I get the error,I use the branch 638.

$ canconvert 2.arxml 1.dbc
ldf is not supported
INFO - convert - Importing 2.arxml ...
Traceback (most recent call last):
  File "D:\soft01\python3.9\Scripts\canconvert-script.py", line 33, in <module>
    sys.exit(load_entry_point('canmatrix==0.9.4.post7+g3a26c02', 'console_scripts', 'canconvert')())
  File "C:\Users\HongZhifu\AppData\Roaming\Python\Python39\site-packages\click\core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "C:\Users\HongZhifu\AppData\Roaming\Python\Python39\site-packages\click\core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "C:\Users\HongZhifu\AppData\Roaming\Python\Python39\site-packages\click\core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Users\HongZhifu\AppData\Roaming\Python\Python39\site-packages\click\core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "D:\soft01\python3.9\lib\site-packages\canmatrix\cli\convert.py", line 153, in cli_convert
    canmatrix.convert.convert(infile, outfile, **options)
  File "D:\soft01\python3.9\lib\site-packages\canmatrix\convert.py", line 68, in convert
    dbs = canmatrix.formats.loadp(infile, **options)
  File "D:\soft01\python3.9\lib\site-packages\canmatrix\formats\__init__.py", line 71, in loadp
    return load(fileObject, import_type, key, **options)
  File "D:\soft01\python3.9\lib\site-packages\canmatrix\formats\__init__.py", line 88, in load
    dbs = module_instance.load(file_object, **options)  # type: ignore
  File "D:\soft01\python3.9\lib\site-packages\canmatrix\formats\arxml.py", line 1959, in load
    result.update(decode_can_helper(ea, float_factory, ignore_cluster_info))
  File "D:\soft01\python3.9\lib\site-packages\canmatrix\formats\arxml.py", line 1876, in decode_can_helper
    sig_value_hash[sig.name] = sig.phys2raw()
  File "D:\soft01\python3.9\lib\site-packages\canmatrix\canmatrix.py", line 442, in phys2raw
    raw_value = (value - self.offset) / self.factor
decimal.InvalidOperation: [<class 'decimal.DivisionUndefined'>]

ALUE, 2022-06-28:
您的邮件已收到！              祝好             刘毅

Wys295723, 2022-09-10:
HI , i have the same problems,could please help me
PS D:\Pycharm\Daily_study\Daily_diag> canconvert D:\Pycharm\Daily_study\Daily_diag\20220729-jh1-ES33_ICM_V4.8.arxml D:\Pycharm\Daily_study\Daily_diag
ldf is not supported
INFO - convert - Importing D:\Pycharm\Daily_study\Daily_diag\20220729-jh1-ES33_ICM_V4.8.arxml ... 
Traceback (most recent call last):
  File "d:\tools\anaconda3\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "d:\tools\anaconda3\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "D:\Tools\Anaconda3\Scripts\canconvert.exe\__main__.py", line 7, in <module>
  File "d:\tools\anaconda3\lib\site-packages\click\core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "d:\tools\anaconda3\lib\site-packages\click\core.py", line 697, in main
    rv = self.invoke(ctx)
  File "d:\tools\anaconda3\lib\site-packages\click\core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "d:\tools\anaconda3\lib\site-packages\click\core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "d:\tools\anaconda3\lib\site-packages\canmatrix\cli\convert.py", line 153, in cli_convert
    canmatrix.convert.convert(infile, outfile, **options)
  File "d:\tools\anaconda3\lib\site-packages\canmatrix\convert.py", line 68, in convert
    dbs = canmatrix.formats.loadp(infile, **options)
  File "d:\tools\anaconda3\lib\site-packages\canmatrix\formats\__init__.py", line 71, in loadp
    return load(fileObject, import_type, key, **options)
  File "d:\tools\anaconda3\lib\site-packages\canmatrix\formats\__init__.py", line 88, in load
    dbs = module_instance.load(file_object, **options)  # type: ignore
  File "d:\tools\anaconda3\lib\site-packages\canmatrix\formats\arxml.py", line 1959, in load
    result.update(decode_can_helper(ea, float_factory, ignore_cluster_info))
  File "d:\tools\anaconda3\lib\site-packages\canmatrix\formats\arxml.py", line 1876, in decode_can_helper
    sig_value_hash[sig.name] = sig.phys2raw()
  File "d:\tools\anaconda3\lib\site-packages\canmatrix\canmatrix.py", line 432, in phys2raw
    raw_value = (value - self.offset) / self.factor
decimal.InvalidOperation: [<class 'decimal.DivisionUndefined'>]

ebroecker, 2022-10-14:
is this issue still there?
Maybe it is fixed already

ebroecker, 2022-12-15:
please reopen with more details if still an issue

closed 2022-12-15 by ebroecker

tedelon, 2023-01-05:
maybe you need  do that:
"pip install ldfparser"