ebroecker/canmatrix #646: no change showed when compare two arxml( i changed the PDU name of arxml)

guohengfeng, 2022-07-22:
when i changed the PDU name in the arxml and tried to compare the difference.   the result didn't show the PDU name change 
 i use cmd like this:cancompare test_bak.arxml test.arxml
i recommand to show the PDU name change  cause when import the arxml to AUTOSAR Tool Chain  the PDU name matters.
[image]]

ebroecker, 2022-07-23:
yes - you are right, a name-change of pdu is not shown.
Background: For now we only use the frame-name and ignore the PDU-name in internal representation.
The communication on CAN does not change, if you change the name in ARXML - but surely this should show up during comparison

ebroecker, 2022-07-23:
Hi @guohengfeng ,

could you try banch `issue_646` ?

`pip install git+https://github.com/ebroecker/canmatrix.git@issue_646`

You need to use the branch above and **additionally** the `-a` switch of cancompare:
`cancompare -a test_bak.arxml test.arxml`

closed 2023-06-22 by ebroecker