ebroecker/canmatrix #656: Missing signal  when excel  convert to dbc

XXIN0, 2022-09-28:
HI,
    During use command line  for excel  convert to dbc,some **signal** will missing.

**_the picture of excel_**
[image]

**_the dbc file_**
[image]

but when I use vector CANdb++ make dbc,then use **canconvert**  for dbc convert to excel，the content of these two copies of Excel is the same, but  excel  convert to dbc the signal will missing that the excel I created by myself ?

_**the two files**_
[input_not_ok_by_make_myself.xlsx](https://github.com/ebroecker/canmatrix/files/9662245/input_not_ok_by_make_myself.xlsx)
[inputok_by_dbc_to_excel.xlsx](https://github.com/ebroecker/canmatrix/files/9662247/inputok_by_dbc_to_excel.xlsx)

ebroecker, 2022-09-28:
Hi @ALUE 

you need to repeat the frame-name in every line:
The working example:
[image]

These repeated frame-names usually have text-color white, thus we don't see it but the have to be there.

BR

XXIN0, 2022-09-29:
> 

HI,ebroecker
  thank you for your advise.

closed 2022-09-29 by XXIN0