ebroecker/canmatrix #661: Default GenSigStartValue autogenerated incorrectly

yesralph, 2022-10-14:
Hi canmatrix,

We recently upgraded our `canmatrix` from `0.9.1` to `0.9.5` and found that the new release autogenerates `GenSigStartValue` based on the signal value range. Below is the minimum reproducible example message and attribute definition:
```
BO_ 560 ECU1_Message: 1 ECU1
 SG_ ECU2_Signal : 0|8@0+ (1,-5) [-2|250] "g" ECU2

BA_DEF_ BO_ "Bus" STRING;
BA_DEF_ SG_ "GenSigStartValue" FLOAT 0.0 100.0;

BA_DEF_DEF_ "Bus" "";
BA_DEF_DEF_ "GenSigStartValue" 10.0;
```

We did not specify the value of `GenSigStartValue` for our signal but did define the default value for `GenSigStartValue` (10.0). When we load the DBC with `canmatrix` and dump the database into a new DBC file, ie.
```
def main():
    db = canmatrix.formats.loadp_flat("in.dbc")
    canmatrix.formats.dumpp({"": db}, "out.dbc")
```

This is what we got
```
BO_ 560 ECU1_Message: 1 ECU1
 SG_ ECU2_Signal : 0|8@0+ (1,-5) [-2|250] "g" ECU2

BA_DEF_ BO_ "Bus" STRING;
BA_DEF_ SG_ "GenSigStartValue" FLOAT 0.0 100.0;
BA_DEF_DEF_ "Bus" "";
BA_DEF_DEF_ "GenSigStartValue" 10.0;

BA_ "GenSigStartValue" SG_ 560 ECU2_Signal 3;
```
The last line was inserted by the `canmatrix`, I believe, because the signal value range `[-2|250]` is narrower than the possible value range that can be generated based on the signal spec `0|8@0+ (1,-5)`, which is [-5|250]`. The autogenerated value `3` is to make sure the starting value is within the specified range: `3*1 - 5 = -2`, even though we already specified the default value for `GenSigStartValue`.

There are two issues IMHO:
1. The specified default value `BA_DEF_DEF_ "GenSigStartValue" 10.0;` is not respected, which potential causes our firmware to start with an unexpected value, ie. `3`.
2. The value range in the signal definition is meant to be equal or narrower than the possible value range the signal can generate. It does not make sense to specify a value range that the signal bits cannot reach, or a value range that matches exactly what the signal bits can generate.

Please help investigate and fix the bug. Thank you so much!

-- Young

ebroecker, 2022-10-16:
I'd assume that this bug was introduced with the merge eb68ec11bab55aaa0922658f9e28455e8bf809f0
(commit was e71416cfb83a499e6883f60c68fe1462b774b93f)

And I think the main problem is, that the default value ist not respected.
what do you think?

yesralph, 2022-10-24:
@ebroecker I think both problems are valid.

yesralph, 2022-10-24:
>And I think the main problem is, that the default value ist not respected.
>what do you think?

Yes.

I think I was confused about how the behavior was triggered. Now I understand the it generates a "safe" default value based on the specified range if no default is provided. However, there seems to be a little inconsistency when it tries to generate the "safe" default value when I tried different configurations.
`SG_ ECU2_Signal : 0|8@0+ (-1,0) [-250|0] "g" ECU2` does not trigger the "safe" default value generation,
but
`SG_ ECU2_Signal : 0|8@0+ (-1,0) [-250|-1] "g" ECU2` generates `BA_ "GenSigStartValue" SG_ 560 ECU2_Signal 250;`

ebroecker, 2022-10-26:
hi @yesralph 

could you try branch `fix_for_661`?

`pip install git+https://github.com/ebroecker/canmatrix.git@fix_for_661`

yesralph, 2022-10-27:
@ebroecker It works fine except when the default value is not provided. Here is the error:
```
    if signal.phys2raw(None) != 0 and float(signal.initial_value) != float(db.signal_defines["GenSigStartValue"].def
aultValue):
TypeError: float() argument must be a string or a number, not 'NoneType'
```

Other places, it's handled with
```
db.signal_defines["GenSigStartValue"].defaultValue is not None
```

Also, it does not automatically generate the "safe" default value anymore. Is that expected?

Thank you for the update!

ebroecker, 2022-10-28:
Hi @yesralph 

Thanks for looking on it.

I updated the branch - maybe I fixed the rest now

closed 2023-07-18 by ebroecker