ebroecker/canmatrix #663: support for pyyaml-6

umlaeute, 2022-10-19:
The `yaml.load()` function (as used in https://github.com/ebroecker/canmatrix/blob/development/src/canmatrix/formats/yaml.py#L77) is inherently unsafe, and therefore the pyyaml developers try to fade it out.
Since pyyaml-5.1 (released on Mar 13, 2019), the module will therefore emit a warning if `yaml.load()` is called (without explictely specifying a `Loader`).
As of pyyaml-6 (released on Oct 13, 2021 - that's about a year ago), the `yaml.load()` function now mandates an explicit *Loader*.

see https://msg.pyyaml.org/load for details.

I currently do not have much time to investigate this issue any further, but the options are:
- ideally use  `yaml.SafeLoader` as the loader (resp. `yaml.safe_load()` which wraps this in a single function) - but this might break with `canmatrix`, as it is building full fledged objects from the yaml-string
- use `yaml.FullLoader` (which basically restores the pre-6.0  behaviour of pyyaml) - but this has to be found to also be unsafe and "may go away".
- do something else.

umlaeute, 2022-10-19:
there's also a bug in the Debian bugtracker to be found at https://bugs.debian.org/1022036

ebroecker, 2022-10-19:
Thanks - should be easy to fix

ebroecker, 2022-10-20:
Hi @umlaeute 

I switched to safe_load()
seems still to work...

umlaeute, 2022-10-20:
cool. i won't be able to do any conclusive testing confirming the "still working", so I'll take your word for it.

ebroecker, 2022-10-26:
OK, I close this ticket.

let's open a new ticket if we get in trouble

closed 2022-10-26 by ebroecker