ebroecker/canmatrix #665: canconvert .dbc to .kcd fails

AKstudios, 2022-10-21:
Trying to convert a dbc file that has scaling, min and max values in the signals. I was able to do this in an older version of canmatrix (>2 years ago), but using the latest commit as of Oct. 20, 2022 gives the following error:

`canconvert source.dbc dest.kcd`
> arxml is not supported
> ldf is not supported
> kcd is not supported
> fibex is not supported
> odx is not supported
> INFO - convert - Importing UPER.dbc ...
> INFO - convert - done
> 
> INFO - convert - Exporting uper.kcd ...
> INFO - convert -
> INFO - convert - 14 Frames found
**> ERROR - __init__ - This file format is not supported for writing**
> INFO - convert - done

ebroecker, 2022-10-23:
I think you need to install lxml library

AKstudios, 2022-10-24:
I already have it installed. When I try to run `pip install lxml`, it gives the following message:

> Requirement already satisfied: lxml in ./.local/lib/python3.9/site-packages (4.9.1)

AKstudios, 2022-11-01:
@ebroecker  when I run `python -m canmatrix.convert`, I get this:

> arxml is not supported
ldf is not supported
kcd is not supported
fibex is not supported
odx is not supported

It does not write anything to file, but it isn't throwing an error either.

ebroecker, 2022-11-03:
I still think, that your python installation does not find the lxml library.

maybe you once more try to install lxml.

AKstudios, 2022-11-04:
Actually, this is completely my fault, I forgot to mention the main thing: I'm running canmatrix on a Raspberry Pi. So what finally worked is installing lxml like so: `sudo apt-get install python3-lxml`.

Thanks for helping out!

closed 2022-11-09 by AKstudios