ebroecker/canmatrix #667: arxml compu parsing problems

danielhrisca, 2022-10-27:
# First issue
The following computation method is not parsed correctly

```
              <COMPU-INTERNAL-TO-PHYS>
                <COMPU-SCALES>
                  <COMPU-SCALE>
                    <LOWER-LIMIT INTERVAL-TYPE="CLOSED">0</LOWER-LIMIT>
                    <UPPER-LIMIT INTERVAL-TYPE="CLOSED">0</UPPER-LIMIT>
                    <COMPU-CONST>
                      <VT>no trailer detected</VT>
                    </COMPU-CONST>
                  </COMPU-SCALE>
                  <COMPU-SCALE>
                    <LOWER-LIMIT INTERVAL-TYPE="CLOSED">1</LOWER-LIMIT>
                    <UPPER-LIMIT INTERVAL-TYPE="CLOSED">1</UPPER-LIMIT>
                    <COMPU-CONST>
                      <VT>trailer detected</VT>
                    </COMPU-CONST>
                  </COMPU-SCALE>
                  <COMPU-SCALE>
                    <SHORT-LABEL>TrailerPresence</SHORT-LABEL>
                    <LOWER-LIMIT INTERVAL-TYPE="CLOSED">0</LOWER-LIMIT>
                    <UPPER-LIMIT INTERVAL-TYPE="CLOSED">0</UPPER-LIMIT>
                    <COMPU-RATIONAL-COEFFS>
                      <COMPU-NUMERATOR>
                        <V>0</V>
                        <V>1</V>
                      </COMPU-NUMERATOR>
                      <COMPU-DENOMINATOR>
                        <V>1</V>
                      </COMPU-DENOMINATOR>
                    </COMPU-RATIONAL-COEFFS>
                  </COMPU-SCALE>
                </COMPU-SCALES>
              </COMPU-INTERNAL-TO-PHYS>
```

`canmatrix` loads this as 

[image]

whereas Vector autosar explorer parses this as:

[image]
[image]

## Second issue
The code [here](https://github.com/ebroecker/canmatrix/blob/92d8c38b03ad7a50cd7e5c0e303c83ca6844cd24/src/canmatrix/formats/arxml.py#L1020) will only parse the value-to-text rule if the upper and lower limit are equal. This means something like this will not work

```
                  <COMPU-SCALE>
                    <LOWER-LIMIT INTERVAL-TYPE="CLOSED">0</LOWER-LIMIT>
                    <UPPER-LIMIT INTERVAL-TYPE="CLOSED">100</UPPER-LIMIT>
                    <COMPU-CONST>
                      <VT>Valid range</VT>
                    </COMPU-CONST>
                  </COMPU-SCALE>
```

ebroecker, 2022-10-28:
Hi @danielhrisca 
thanks for reporting this issue.

I tried to fix the fist issue in branch `feature_667`

second issue is more complicated.
Not sure how to do this elegant in canmatrix, because we for now have no option to store value-tables with ranges

ebroecker, 2024-12-18:
hi,
long time ago...

I just added something like this for LDF:

https://github.com/ebroecker/canmatrix/pull/823

ebroecker, 2026-11-16:
should be fixed with #943

closed 2026-11-16 by ebroecker