ebroecker/canmatrix #670: canconvert *.arxml *.dbc issue

Mattas11, 2022-11-10:
Hi,
I try to convert arxml to dbc, but I am getting following error:

> C:\Users\DoesNotMatter>canconvert C:\Users\DoesNotMatter\Desktop\canmatrix_test\Any.arxml C:\Users\DoesNotMatter\Desktop\canmatrix_test\Any.dbc
> xls is not supported
> INFO - convert - Importing C:\Users\DoesNotMatter\Desktop\canmatrix_test\Any.arxml ...
> Traceback (most recent call last):
>   File "C:\Users\DoesNotMatter\AppData\Local\Programs\Python\Python310\lib\runpy.py", line 196, in _run_module_as_main
>     return _run_code(code, main_globals, None,
>   File "C:\Users\DoesNotMatter\AppData\Local\Programs\Python\Python310\lib\runpy.py", line 86, in _run_code
>     exec(code, run_globals)
>   File "C:\Users\DoesNotMatter\AppData\Local\Programs\Python\Python310\Scripts\canconvert.exe\__main__.py", line 7, in <module>
>   File "C:\Users\DoesNotMatter\AppData\Local\Programs\Python\Python310\lib\site-packages\click\core.py", line 1130, in __call__
>     return self.main(*args, **kwargs)
>   File "C:\Users\DoesNotMatter\AppData\Local\Programs\Python\Python310\lib\site-packages\click\core.py", line 1055, in main
>     rv = self.invoke(ctx)
>   File "C:\Users\DoesNotMatter\AppData\Local\Programs\Python\Python310\lib\site-packages\click\core.py", line 1404, in invoke
>     return ctx.invoke(self.callback, **ctx.params)
>   File "C:\Users\DoesNotMatter\AppData\Local\Programs\Python\Python310\lib\site-packages\click\core.py", line 760, in invoke
>     return __callback(*args, **kwargs)
>   File "C:\Users\DoesNotMatter\AppData\Local\Programs\Python\Python310\lib\site-packages\canmatrix\cli\convert.py", line 153, in cli_convert
>     canmatrix.convert.convert(infile, outfile, **options)
>   File "C:\Users\DoesNotMatter\AppData\Local\Programs\Python\Python310\lib\site-packages\canmatrix\convert.py", line 68, in convert
>     dbs = canmatrix.formats.loadp(infile, **options)
>   File "C:\Users\DoesNotMatter\AppData\Local\Programs\Python\Python310\lib\site-packages\canmatrix\formats\__init__.py", line 71, in loadp
>     return load(fileObject, import_type, key, **options)
>   File "C:\Users\DoesNotMatter\AppData\Local\Programs\Python\Python310\lib\site-packages\canmatrix\formats\__init__.py", line 88, in load
>     dbs = module_instance.load(file_object, **options)  # type: ignore
>   File "C:\Users\DoesNotMatter\AppData\Local\Programs\Python\Python310\lib\site-packages\canmatrix\formats\arxml.py", line 1959, in load
>     result.update(decode_can_helper(ea, float_factory, ignore_cluster_info))
>   File "C:\Users\DoesNotMatter\AppData\Local\Programs\Python\Python310\lib\site-packages\canmatrix\formats\arxml.py", line 1852, in decode_can_helper
>     frame = get_frame(frameTrig, ea, multiplex_translation, float_factory, headers_are_littleendian)
>   File "C:\Users\DoesNotMatter\AppData\Local\Programs\Python\Python310\lib\site-packages\canmatrix\formats\arxml.py", line 1595, in get_frame
>     get_signals(pdu_sig_mapping, new_frame, ea, None, float_factory)
>   File "C:\Users\DoesNotMatter\AppData\Local\Programs\Python\Python310\lib\site-packages\canmatrix\formats\arxml.py", line 1103, in get_signals
>     get_signalgrp_and_signals(isignal, isignal_array, frame, group_id, ea)
>   File "C:\Users\DoesNotMatter\AppData\Local\Programs\Python\Python310\lib\site-packages\canmatrix\formats\arxml.py", line 989, in get_signalgrp_and_signals
>     e2e_transform['data_ids'] = [int(x.text) for x in data_id_elems]
>   File "C:\Users\DoesNotMatter\AppData\Local\Programs\Python\Python310\lib\site-packages\canmatrix\formats\arxml.py", line 989, in <listcomp>
>     e2e_transform['data_ids'] = [int(x.text) for x in data_id_elems]
> ValueError: invalid literal for int() with base 10: '0xa3'

Another issue I found:
When I tried

> C:\Users\DoesNotMatter>canconvert C:\Users\DoesNotMatter\AppData\Local\Programs\Python\Python310\Lib\site-packages\canmatrix\tests\ARXML_min_max.arxml C:\Users\DoesNotMatter\AppData\Local\Programs\Python\Python310\Lib\site-packages\canmatrix\tests\ARXML_min_max_test.dbc
> xls is not supported
> INFO - convert - Importing C:\Users\DoesNotMatter\AppData\Local\Programs\Python\Python310\Lib\site-packages\canmatrix\tests\ARXML_min_max.arxml ...
> WARNING - arxml - ARXML signal scaling: polynom is not supported and it is replaced by factor=1 and offset =0.
> WARNING - arxml - ARXML signal scaling: polynom is not supported and it is replaced by factor=1 and offset =0.
> WARNING - arxml - ARXML signal scaling: polynom is not supported and it is replaced by factor=1 and offset =0.
> INFO - convert - done
> 
> INFO - convert - Exporting C:\Users\maasma7\AppData\Local\Programs\Python\Python310\Lib\site-packages\canmatrix\tests\ARXML_min_max_test.dbc ...
> INFO - convert - New_CanCluster
> INFO - convert - 1 Frames found
> INFO - convert - done

it converted, but not with the correct Start-Positions/Startbits (see attached pics).
[image]
[image]

ebroecker, 2022-11-21:
Hi @Mattas11 
which version do you use?

At least the first Error was fixed already: https://github.com/ebroecker/canmatrix/commit/97ebe53c66556410d9484f7335c6702f24c19b5f

Mattas11, 2022-11-22:
Thanks for response and sry, that I did not find it.

Version I use:
Python: 3.10.0
canmatrix: 0.9.5

I will soon try the fixes from your link.

Mattas11, 2022-11-22:
Hi @ebroecker!

With the fix I'm now available to convert *.arxml to *.dbc, but how can I be sure, that the conversion worked if the sample file I mentioned still does not convert correctly?

Thanks and best regards!