ebroecker/canmatrix #671: Incorrect handling of value tables in phys2raw and raw2phys

larslj, 2022-11-16:
The value tables in a DBC file stores raw values and the corresponding string. The methods phys2raw() and raw2phys() are written as if the physical value is stored in the value table.

E.g., phys2raw('NotAvailable') should return 0xFF on a typical 8-bit J1939 signal, regardless of offset and factor.

ebroecker, 2022-12-13:
Hi @larslj 

Just to be sure, I will fix the right thing....

Does is this problem only exist for "named" values (from valuetabe), right?
The value in the value-table is already *raw* - canmatrix for now does scale the already raw value once more as if it was physical, right?

Thanks in advance

larslj, 2022-12-13:
Yes, this is only for the value table strings. Numerical arguments to these functions are handled correctly. Thanks for looking in to this.

ebroecker, 2022-12-15:
hi @larslj 

could you try branch issue_671?

pip install git+https://github.com/ebroecker/canmatrix.git@issue_671

thanks

larslj, 2022-12-16:
Yes, I can confirm that the branch works as expected. Thank you.

commit 7332f914 (2022-12-16): fix for #671 (#673)

value-tables contain raw values

fixes #671
closed 2022-12-16 by ebroecker