ebroecker/canmatrix #672: Multiple Ecus arxml to dbc

fogmercury, 2022-12-02:
Hi all,
I have a problem when converting an arxml file with multiple Ecus but same cluster to a dbc file
Used Version: can matrix 0.9.5
Example Data
[TestCluster.zip](https://github.com/ebroecker/canmatrix/files/10141295/TestCluster.zip)
Used Command 
'canconvert -vv  TestCluster.arxml TestCluster.dbc'

Expected Behavior:
Two Ecus are present in DBC file

Seen Behavior
Only one Ecu is present.
Seems that only the first Ecu is exported

Best regards

fogmercury, 2023-01-02:
Hi @ebroecker 
better but not fully right yet. 
Tx Looks fine
The RX Signal is not mapped correctly.
I added an expected file 
[aa_New_CanCluster_fixed.zip](https://github.com/ebroecker/canmatrix/files/10331668/aa_New_CanCluster_fixed.zip)

ALUE, 2023-01-02:
您的邮件已收到！              祝好             刘毅

ebroecker, 2023-01-10:
Hi  @fogmercury 

could you try branch `issue_672`?
pip install git+https://github.com/ebroecker/canmatrix.git@issue_672

Thanks

fogmercury, 2023-01-24:
Hi ebroeker,
sorry for the late reply. Looks good so far. 
Thanks for your quick support.

ebroecker, 2023-01-24:
Thanks - I merged the corresponding branch.
I'll close this issue here, please open a new issue, if something still goes wrong