ebroecker/canmatrix #685: Reassigning Signal's Startbits after other Signals Deletion

aku666, 2023-02-10:
Sample DBC's Message has 4 Signals as shown :
[image]

After deleting Sample_Signal & Sample_Signal3, it looks like :
[image]

Is it possible to reassign the startbits of remaining signals, i.e., making startbit of Sample_Signal2 as 0 instead of 16 and similarly for Sample_Signal4's startbit as 16 instead of 48?
If the above operation is possible can I get an explanation as to how I can execute it?

ebroecker, 2023-02-12:
Hi @aku666 

maybe I could create some command line switch like "--remove_gaps" ?
Would that help?

aku666, 2023-02-12:
Hi @ebroecker,

Thank you & yes, that would help me a lot and others as well who wanted/or might want this feature in the future.

But just out of curiosity, if I may ask, why this feature was not thought of previously or was it in your mind but did not implement due to any reason?

ebroecker, 2023-02-13:
Hi @aku666 

it sounds really useful but nobody asked for that feature till now...

I did a first implementation:
could you try branch frature_frame_compress_685 ?

`pip install git+https://github.com/ebroecker/canmatrix.git@frature_frame_compress_685`

P.S. Yes - there's a typo in the banch-name :-)

aku666, 2023-02-13:
Hi @ebroecker,
Thanks a lot. I'll check this branch and revert back🙂

aku666, 2023-02-14:
Hi @ebroecker, 
The modifications are working exactly as expected.
Thank you.

ebroecker, 2023-02-14:
OK,
@aku666 
thanks for the fast reply!
I'll merge these days

closed 2023-06-20 by ebroecker