ebroecker/canmatrix #686: Names of the excel column header changing according to the can dbc for better import and understanding

lushee, 2023-02-10:
Hi @ebroecker sir big fan of your work ,
These  are the issues I observed during using of tool
1)The names of the column headers are different from the CAN dbc matrix so would like to suggest to changes it according to can DBC columns for better and simpil export 

ex :
[image]

here the naming notation is standard so different columns for direction (Tx , Rx ), Maximum , minimum values and other proper naming notations will be greatly useful. 

2)Most of the time the next message-id is not being recognized unless each signal is named with the  message id repetitively 

3)Incase of multiple sheets it will be useful to add a flag for a specific sheet in xls , here the xlsx format is not being supported 

If I have spoken immaturely please pardon me sir ,
Thank you ,
Regards

ebroecker, 2023-02-12:
Hi @lushee ,

Thanks for your input.

Well - I have quite a bit another view...

1+2: You are right - candb++ is the most used tool for dbc. But for canmatrix dbc is only one from many formats.
Excel-Format in canmatrix is far from satisfactory but I do not have a complete picture how to make it really good.
Expecially excel is a lossy format - you will loose a lot of details if you do canmatrix->xlsx->canmatrix:
Thus I wish a *complete* excel-format which also is editable by hand - But I don't habe an idea how this could work.

3: this should not be to complicated - I can have a look on it.

lushee, 2023-02-12:
Hi , @ebroecker 
Thank you for swift reply sir , The start bit , value tables are the parts where its confusing for me , i saw in the code that there is a formula for calculating start bit something like ((signal byte *factor)-8 + signal bit )( ps:i may be wrong ) is there any way to provide start bit directly in excel column cause most OEM'S give the 'startbit ' in can matrix if not a simple excel formula can do the trick --- so the direct start bit as input will be hassle free , also the value table import I am still confused.

Regards.

ebroecker, 2023-02-13:
Hi @lushee ,

the startbit is a bit an issue.
For little-endian signals the starbit in excel is 1:1 the one we use internal.

The crap begins with the big-endian/motorola/MOST-SIGNIFICANT-BYTE-FIRST signals:
You can have different startbit-notations for these motorola-signals...
But you can control this:

Please search in the documentation
https://canmatrix.readthedocs.io/en/latest/cli.html

for the command-line-switch:
`--xlsMotorolaBitFormat`

Something about signal-byte-orders:
https://github.com/ebroecker/canmatrix/wiki/signal-Byteorder

Hope that helps
P.S.
"Startbit Directly in Excel": We have different startbit-notations in different fileformats, thus "direct" is not really clear what it means :-)

ebroecker, 2023-07-18:
Did not hear any commnet, so I'll close this issue.

If you have still further questions please open a new issue

closed 2023-07-18 by ebroecker