ebroecker/canmatrix #688: Gateway info is not supported

AhmedKotb10, 2023-02-26:
AR-PACKAGE can have a gateway mapping info. This info is not supported by canmatrix so far.
is there a way to get these info or is it planned to be supported by canmatrix?

ebroecker, 2023-02-26:
Hi @AhmedKotb10 

there are many hings in ARXML we do not (yet) support.

It would be relative easy to read the gateway info. But what to do with this information?

I know no other format that supports this info.

AhmedKotb10, 2023-02-28:
Hi,
I'm concerned about gateway info as I need this info for some testing automation stuff. Therefore, I though it would be a nice enhancement to have this info in this library rather than using other libraries or methods to get this info.

ebroecker, 2023-03-02:
ok, hm, i understand a bit better now.

would the pdu-mapping (pduX from bus to bus) enough for you?

AhmedKotb10, 2023-03-05:
That will be even great. For me, I used to parse also I-SIGNAL-MAPPING info (signalX of pduX from busX to signalY of pduY on busY)

ebroecker, 2023-03-05:
Hi @AhmedKotb10 ,

I'll start to implement the extraction of the pdu-mapping and the signal-mapping.
This should be part of the can-cluster-information. 
We'll see how it could be stored somehow logical...

ebroecker, 2023-03-05:
Hi @AhmedKotb10 

I did a first (very basic) implementation:

branch: https://github.com/ebroecker/canmatrix/tree/cr_cluster_routing_info

PR: https://github.com/ebroecker/canmatrix/pull/692

have a look at https://github.com/ebroecker/canmatrix/blob/458c8168d0da795e4b424d51d3afdd2cc319d043/src/canmatrix/tests/test_arxml_gw.py#L95 

or use `cluster.pdu_gateway()` to get complete pdu-routing info.

You can directly try it:
`pip install git+https://github.com/ebroecker/canmatrix.git@cr_cluster_routing_info`

Please give some feedback, if this goes into the right direction and is useful for you?
After that, I'll implement the same for signals also.

ebroecker, 2023-03-12:
@AhmedKotb10 

Hi, did you have a chance to look on the implementation?

AhmedKotb10, 2023-03-14:
Hi Bröcker,
I would like to thank you for your efforts. Unfortunately, I didn't have
time to test it so far, I will try to check it during next weekend and give
you feedback.

AhmedKotb10, 2023-03-20:
Hello,
Nice job!. It is working so good. 
Suggestions for what can be added:
return routing info with following data ("pdu_name", "bus_name", "pdu_type")
- pdu_type --> {DCM-I-PDU, I-SIGNAL-I-PDU, NM-PDU ...etc}, this is used -in my case- for example to distinguish between diagnostics frames routed and normal frames

ebroecker, 2023-03-20:
Hi @AhmedKotb10,

thanks for your feedback.

Gonna see what I can do. Stay tuned.

ebroecker, 2023-03-20:
Hi @AhmedKotb10 

I added some more gateway-info now

ebroecker, 2023-04-25:
ok, no answers, gonna merge the pr

closed 2023-04-25 by ebroecker