ebroecker/canmatrix #694: can't convert xlsx into canfd dbc

XXIN0, 2023-03-27:
hello,
I can't convert xlsx into canfd dbc
here is my problem discribe
step:
1、edit and save an canfd template dbc file by candb++ ( name: "demofd.dbc")
2、use command line **_“canconvert  demofd.dbc demofd.xlsx"_**
[image]

3、edit  demofd.xlsx，then use command line  **_”canconvert  demofd.xlsx new_demofd.dbc"_** 

result:
the new_demofd.dbc is not canfd message type.
expect:
the new_demofd.dbc is canfd message type.

ebroecker, 2023-03-27:
Hi  @XXIN0 ,

Short answer: Works as specified :-)

Long answer:

Excel is (as documented) a **lossy format**; 

If you just want to work-around the fd-issue you can do something like:
`canconvert --additionalFrameAttributes=is_fd some.dbc some.xlsx`

XXIN0, 2023-03-27:
> Hi @XXIN0 ,
> 
> Short answer: Works as specified :-)
> 
> Long answer:
> 
> Excel is (as documented) a **lossy format**;
> 
> If you just want to work-around the fd-issue you can do something like: `canconvert --additionalFrameAttributes=is_fd some.dbc some.xlsx`

> thank you,I'm sorry  I'm not look at the documentation in detail.

closed 2023-03-27 by XXIN0