ebroecker/canmatrix #697: Partial conversion of arxml to dbc

ApoorvaSatish, 2023-04-12:
Hi,
I want to convert arxml to dbc

I received the below output.

C:\..\convert.py:293: SyntaxWarning: "is" with a literal. Did you mean "=="?
  if len(frame.transmitters) is 0:
C:\..\convert.py:300: SyntaxWarning: "is" with a literal. Did you mean "=="?
  if (signal.phys2raw(signal.max) - signal.phys2raw(signal.min)) is 0:
C:\..\convert.py:307: SyntaxWarning: "is" with a literal. Did you mean "=="?
  if signal.unit is "" and len(signal.values) == 0:
Traceback (most recent call last):
  File "C:\..\convert.py", line 24, in <module>
    import copy
  File "C:\..\copy.py", line 25, in <module>
    import logging
  File "C:\Program Files\Python39\lib\logging\__init__.py", line 26, in <module>
    import sys, os, time, io, re, traceback, warnings, weakref, collections.abc
  File "C:\Program Files\Python39\lib\re.py", line 124, in <module>
    import enum
  File "C:\Program Files\Python39\lib\enum.py", line 2, in <module>
    from types import MappingProxyType, DynamicClassAttribute
  File "C:\..\types.py", line 3, in <module>
    import typing
  File "C:\Program Files\Python39\lib\typing.py", line 23, in <module>
    import contextlib
  File "C:\Program Files\Python39\lib\contextlib.py", line 6, in <module>
    from functools import wraps
  File "C:\Program Files\Python39\lib\functools.py", line 22, in <module>
    from types import GenericAlias
ImportError: cannot import name 'GenericAlias' from partially initialized module 'types' (most likely due to a circular import) (C:\..\types.py)

Can you please help me?

ebroecker, 2023-04-12:
which version do you use?

could you use recent canmatrix vom development branch

ebroecker, 2023-04-13:
Hi @ApoorvaSatish ,

I guess you cannot sent this arxml to me?

could you convert with -vv option? This will generate a lot of debug-output.

AND - could you please try the current version from developement-branch.

BR
Eduard

ebroecker, 2023-06-20:
Got no further information...

please reopen if more information available

closed 2023-06-20 by ebroecker