ebroecker/canmatrix #704: canconvert doesn't extract value table of a multiplexor signal

bomm, 2023-06-02:
I installed canmatrix on a Windows system using `pip install canmatrix`. `pip list` shows `canmatrix 0.9.5`. 
Python 3.7.3

I created an example file `test3.dbc` using Vector CANdb++.

Message `New_Message_2` has value tables both for the multiplexor signal `New_Signal_2` and for the multiplexed signal `New_Signal_3`.

When I extract this message with `canconvert --frames=New_Message_2 test.dbc msg2.dbc`, the output file contains the value table for the multiplexed signal `New_Signal_3` but not for the multiplexor signal `New_Signal_2`.

input file [test.dbc.txt](https://github.com/ebroecker/canmatrix/files/11634207/test.dbc.txt)
resulting output file [msg2.dbc.txt](https://github.com/ebroecker/canmatrix/files/11634209/msg2.dbc.txt)

bomm, 2023-06-02:
In `src/canmatrix/formats/dbc.py`, function `def dump(in_db, f, **options):`, there is the following code to write value tables:
```
    # signal-values:
    for frame in db.frames:
        multiplex_written = False
        for signal in frame.signals:
            if signal.multiplex == 'Multiplexor' and multiplex_written:
                continue

            multiplex_written = True

            if signal.values:
                f.write(
                    ('VAL_ %d ' %
                     frame.arbitration_id.to_compound_integer() +
                     output_names[frame][signal]).encode(dbc_export_encoding, ignore_encoding_errors))
                for attr_name, val in sorted(signal.values.items(), key=lambda x: int(x[0])):
                    if '"' in val:
                        val = val.replace('"', '\\"')
                    f.write(
                        (' ' + str(attr_name) + ' "' + val + '"').encode(dbc_export_encoding, ignore_encoding_errors))

                f.write(";\n".encode(dbc_export_encoding, ignore_encoding_errors))
```

When I comment the lines
~~~
            if signal.multiplex == 'Multiplexor' and multiplex_written:
                continue
~~~
it seems to work with my DBC files.  

I don't understand the purpose of the condition. As I understand the code it will skip writing the value table for Multiplexor signals except if it is processed in the first loop iteration.
I don't know if there are any use cases where this change would have any bad effect.

ebroecker, 2023-06-20:
Hi  @bomm 

thanks for this report.

I think this was a copy-paste error, but no idea...

could you try branch issue_704
which more or less does the same you did with commenting out...

pip install git+https://github.com/ebroecker/canmatrix.git@issue_704

Thanks

closed 2023-07-18 by ebroecker